FloatingArrayDesign/MoorDyn #3: Problem with MoorDyn in Linux OS

AB1982, 2021-04-25:
I m trying to use the MoorDyn library with matlab, installed on a LINUX machine. When I try to load library I get an error because the library are looking for the file Windows.h. Looking inside the file MoorDyn.h, if I well understand, it looks for a variable MoorDyn_EXPORTS that shoud be set to LINUX. I try to set it with the setenv command in matlab but it does not work, and also I try to set that variable before compiling the library but the results does not change. 
Have you any suggestions to overcame that issue? Thanks

naugler, 2021-05-21:
The MoorDyn.h file checks to see if "OSX" is NOT DEFINED and "LINUX" is NOT DEFINED. (Those two flags are defined in the compileDYLIB/makefile and the compileSO/makefile, respectively.) The #include <Windows.h> line should only be considered when those flags are both not defined, which happens when you are building using compileDLL/makefile on a Windows machine.

Make sure you are using the correct makefile to build the appropriate libraries for your system.
compileDLL/makefile = Windows
compileDYLIB/makefile = OSX
compileSO/makefile = Linux

AB1982, 2021-05-23:
Thank for the suggestion, but I still cannot solve the problem. I used the proper makefile (compileSO/makefile) and the compiling process works fine, but when I try to used the library, it still looking for  Windows.h. If I well understand, the flags you refer to is -DLINUX for linux and -DOSX for mac inserted at the line CFLAGS in the makefile, respectively? If it is so, it seems that it does not recognize such flag (at least for linux)

mattEhall, 2021-05-23:
Hi @AB1982, I'm inexperienced with Linux but will try to help if I can. Could you clarify what version of MoorDyn you're working with, and how you're compiling it?
As you note, Windows.h is included (perhaps unnecessarily) in MoorDyn.h, which Matlab references when loading the MoorDyn library. I wonder if Matlab might not support preprocessor directives in this situation. Have you tried commenting out the line that imports Windows.h?

AB1982, 2021-05-24:
Hi @mattEhall , I m using MoorDyn 1.01.01C, As you suggest I recompiled the library commenting the lines concering the call to windows.h in the MoorDyn.h file:
``
/*
#ifndef OSX
#ifndef LINUX
 #include <Windows.h>
#endif
#endif
*/
``
In this way I can load the library and the matlab libfunction subroutine shows all the functions contained in the library.
By this way everythings seems to be fine..

closed 2021-06-12 by AB1982