FloatingArrayDesign/MoorDyn #4: can you share your test model of your paper?

linmeihong, 2021-05-21:
Can you share your test model data of your paper?
M. Hall and A. Goupee, “Validation of a lumped-mass mooring line model with DeepCwind semisubmersible model test data,” Ocean Engineering, vol. 104, pp. 590–603, Aug. 2015.

I just want to test the code by the test model.

mattEhall, 2021-05-23:
Some of the input files I used aren't mine to share, but I can share my MoorDyn input file and point you to some generic input files for the design here: https://github.com/OpenFAST/r-test/tree/main/glue-codes/openfast/5MW_OC4Semi_WSt_WavesWN

Below is the input file I used in the paper (it was for the first version of MoorDyn so may need some adjustments depending on what version you have):

> MoorDyn input file for 2011 DeepCwind Semi tests - mhall - tuned line lengths Sept 2014
> ---------------------- LINE DICTIONARY -----------------------------------------------------
> LineType  Diam    MassDenInAir    EA        cIntDamp     Can     Cat    Cdn     Cdt   
> (-)       (m)       (kg/m)        (N)        (Pa-s)      (-)     (-)    (-)     (-)  
> main     0.13376    116.62      753.6E6       1.0e8     0.865   0.269  1.08   0.213      
> ---------------------- NODE PROPERTIES -----------------------------------------------------
> Node      Type      X        Y         Z        M        V        FX       FY      FZ
> (-)       (-)      (m)      (m)       (m)      (kg)     (m^3)    (kN)     (kN)    (kN)
> 1         Fix    -837.6       0.0    -200       0        0        0        0       0
> 2         Fix     418.8    -725.383  -200       0        0        0        0       0
> 3         Fix     418.8     725.383  -200       0        0        0        0       0
> 4         Vessel  -40.868     0.0     -14       0        0        0        0       0
> 5         Vessel   20.434   -35.393   -14       0        0        0        0       0
> 6         Vessel   20.434    35.393   -14       0        0        0        0       0
> ---------------------- LINE PROPERTIES -----------------------------------------------------
> Line     LineType  UnstrLen  NumNodes   NodeAnch  NodeFair  Flags/Outputs
> (-)      (-)       (m)         (-)       (-)       (-)       (-)
> 1        main     833.6         20        1         4        -
> 2        main     834.8         20        2         5        -
> 3        main     834.85        20        3         6        -
> ---------------------- SOLVER OPTIONS-----------------------------------------
> 0.001    DT 		- time step to use in mooring integration
> 0.05     DWWave 	- wave d omega to use in compressed wave calculations
> 3.0e6    kb                     - bottom stiffness
> 3.0e5    cb                     - bottom damping
> 0        WaveKin                - wave kinematics flag (1=include, 0=neglect)
> 200      WtrDpth                - water depth - this should be handled through FAST!
> --------------------- need this line ------------------

linmeihong, 2021-05-24:
Thank you very much. I want to rebuild the code in java language, I just want the basic information of the model to compute another way.  if I succeed, I will reference to your paper. 
I'll ask you for further advice later for Information which i can not found.Athank you v ery  much.

linmeihong, 2021-06-09:
dear Matthew Hall,
thank you very much, .can you privde the source test data of  <fig.9. fairlead tensions in regular wave 5> to me? I just use as a comparison. from your instruction, i'll create the model and make a comparison without coupled floating wind turbine model validation. thank you very much. <linmeihong@qq.com>

linmeihong, 2021-06-19:
dear Matthew Hall,
     may be your have difficult for my question, thank you very mush for your reply. some of the computation parameters (figure 6 wave 5) in your paper  I still not understand:
1. such as the <Table4 Platform quadraticdragcoefficients>, just add to the  C matrix of motion equation equation(M+m)X'' + CX' + KX = F(t)?
2. In the motion equation m is m of infinite frequency or just m(w) where w is omega of wave? , C is C(w) damping of wave omega?
Can you give me som instructions? thank you very much.

mattEhall, 2021-06-19:
Hi, sorry for missing your previous message. Much of the data used in my paper belongs to others, and it was for very specific tests, so it's not ideal for sharing. If you need exact data to compare against, I'd suggest using the data in the r-test repository that I shared the link for previously. That data is all in a consistent format from OpenFAST so it is easier to work with.

I don't really understand your two new questions or what they are referring to, sorry. In case it helps, the coefficients mentioned in Figure 6 correspond to those in equations 8-11. And the mass and damping matrices in MoorDyn come from Morison's equation, as the equations show, so MoorDyn does not ever have mass or damping matrices that are expressed as a function of frequency.

linmeihong, 2021-06-20:
thank you Matthew Hall. I do not use openfast, I rebuild the oc4 model in aqwa, and implement the floating structure motion equation myself, I use the mass and Ixx,Iyy,Izz(table3) in your paper, and my purpose is just use the parameters of your paper. I only want to use the figure6 for comparison. more comparison may do as your instruction in r-test respository.  I have no experience in openfast now. my be I learn it soon, but now i want to test floating structure motion equation and mooring system which is implemented the same as your paper. 
the two problems are for the computation of motion equation of oc4 floating structure.
1. such as the damping parameters(Table4 Platform quadratic drag coefficients.), just add to the C matrix of motion equation equation(M+m)X'' + CX' + KX  + Integral(R(t-\tau)X')d\tau= F(t)?
2. In the motion equation of OC4 , m is m of infinite frequency or just m(w) where w is omega of wave? , C is C(w) damping of wave omega? wave5 is regular wave.
3. does your computation use the delay function(velocity impulse function matrix) for damping computaion?
Can you give me som instructions? 
thank you very much. I have difficult in the test of the model(motion equation of oc4 floating structure and mooring system) because there is less papers that contain Sufficient calculation parameters. if I succeed, I just want the computation data in your paper by enlarging the image of figure6,  without using the data for very specific tests as you mention. 
thank s a lot.

djanders19, 2022-07-22:
Hi Matt, reviving this old thread to ask if you'd be able to share the Moordyn input files from [this v2 paper](https://www.nrel.gov/docs/fy20osti/76555.pdf) - I'd be particularly interested in the inputs used to generate fig 13. I've been trying to link lines together with a massless connection node as described in the paper - however, whenever I do so the connection node does not seem to move.

For reference, the input file I'm using looks like:

> --------------------- MoorDyn Input File -------------------------------------------------------
> Lorem ipsum dolor
> ----------------------- LINE TYPES --------------------------------------------------------------
> TypeName   Diam    Mass/m     EA         BA/-zeta    EI         Cd     Ca     CdAx    CaAx
> (name)     (m)     (kg/m)     (N)        (N-s/-)     (N-m^2)    (-)    (-)    (-)     (-)
> chain       13.332E-3  1.1        7.51E6     -0.5        0        0.0    0.0      0.0     0.0
> ---------------------------- BODIES --------------------------------------------------------------
> ID   Attachment  X0     Y0    Z0     r0      p0     y0     Mass  CG*   I*      Volume   CdA*   Ca
> (#)     (-)      (m)    (m)   (m)   (deg)   (deg)  (deg)   (kg)  (m)  (kg-m^2)  (m^3)   (m^2)  (-)
> ----------------------- POINTS -------------------------------------------------------------------
> Node      Type      X        Y         Z        M        V         CdA   CA
> (-)       (-)      (m)      (m)       (m)      (kg)     (m^3)     (m^2)  (-)
> 1        Fixed   15.0       0.0      -16.6       0       0         0      0
> 2        Connect 15.0       4.0      -16.6       0       0         0      0
> 3        Fixed   15.0       8.0      -16.6       0       0         0      0
> -------------------------- LINES -----------------------------------------------------------------
> Line     LineType NodeA     NodeB  UnstrLen  NumSegs     Flags/Outputs
> (-)      (-)       (-)       (-)   (m)         (-)          (-)
> 1 chain 1 2 5 5 p
> 2 chain 2 3 5 5 p
> -------------------------- SOLVER OPTIONS----------------------------------------------------------
> 2        writeLog     - Write a log file
> 0.000220 dtM          - time step to use in mooring integration
> 3.0e6    kb           - bottom stiffness
> 3.0e5    cb           - bottom damping
> 4000    WtrDpth      - water depth
> 4.0      ICDfac       - factor by which to scale drag coefficients during dynamic relaxation IC gen
> 0.0000015    threshIC     - threshold for IC convergence
> 1.0    TmaxIC       - threshold for IC convergence
> 0.00001     dtIC         - Time lapse between convergence tests (s)
> ------------------------- need this line -------------------------------------- 

And the output looks like this after 16s of integration:
[image]

I've noticed something similar with 'Free' type points - I'd assumed that they are supposed to allow line ends to move freely, but in fact they seem to stay fixed in space. 

If you'd like, I can open up another Issue thread for this - I just wanted to be sure the input syntax was correct before filing it as an issue, hence the request for the paper input.

djanders19, 2022-08-10:
Replying to myself here in case others have the above issue: I was mistakenly using the dev branch as opposed to the v2 branch, where Matt pushed a commit fixing the behavior of connect/free points - see [this commit](https://github.com/mattEhall/MoorDyn/commit/7f3485315a1fcd806e318eb062a370da70c77980).

sanguinariojoe, 2023-01-07:
Can we close this?

linmeihong, 2023-01-09:
yes, thank you very much.

closed 2023-03-24 by sanguinariojoe