FloatingArrayDesign/MoorDyn #5: can you provide the sample "python script" mentioned in the documentation?

yonghoonlee, 2021-05-24:
I would like to utilize the specific version of MoorDyn (1.01.01C) with python.
However, I could not find any example to call MoorDyn from python,
but only a short note on the documentation saying "MoorDyn can be called from Python scripts."
Would you be able to provide me with the sample python script?
Thank you.

trashid10, 2021-05-25:
I have been using the ctypes library in Python to interface with MoorDyn (pre-compiled DLL) and I more or less followed the MATLAB example to get the ball rolling.

I haven't been able to to get the tension to change over time however, so perhaps a detail is missing between the MATLAB->Python translation. I used the example lines.txt file provided in the user guide, and my tension is a constant 8.29112717e+08.

mattEhall, 2021-06-09:
Hi @yonghoonlee , sorry for the slow response here. Below is an example. It would need some adjustments depending on the version, but this should give the general idea (if you haven't done it already).

@trashid10, I recall having a similar issue as what you describe. When I use MoorDyn from Python I just read the MoorDyn output files after the fact rather than getting tensions from it at each time step, so I've avoided that issue so far. It would be nice to figure out a solution though.

```
import numpy as np
from ctypes import *

nFair = 2     # number of fairleads (Vessel nodes in MoorDyn)

# make MoorDyn function prototypes, parameter lists; load it and setup functions
MDInitProto = CFUNCTYPE(c_int, c_double*6, c_double*6)  # remember, first entry is return type, rest are args
MDCalcProto = CFUNCTYPE(c_int, c_double*6, c_double*6, c_double*6, c_double*1, c_double*1)
MDFairCalcProto = CFUNCTYPE(c_int, c_double*(nFair*3), c_double*(nFair*3), c_double*(nFair*3), c_double*1, c_double*1)
MDConnProto = CFUNCTYPE(c_int, c_int, c_double*3)
MDFairProto = CFUNCTYPE(c_int, c_int, c_double*3)
MDClosProto = CFUNCTYPE(c_int)

MDInitParams = (1, "x"), (1, "xd")
MDCalcParams = (1, "x"), (1, "xd"), (2, "f"), (1, "t"), (1, "dtC")  # don't call this with the output parameter, it will be returned instead
MDFairCalcParams = (1, "x"), (1, "xd"), (2, "f"), (1, "t"), (1, "dtC") 
MDConnParams = (1, "l"), (2, "r")
MDFairParams = (1, "l"), (2, "f")

MDdll = WinDLL("MoorDyn.dll")   # load MoorDyn DLL

MDInit = MDInitProto(("LinesInit", MDdll), MDInitParams)
MDCalc = MDCalcProto(("LinesCalc", MDdll), MDCalcParams)
MDFairCalc = MDFairCalcProto(("FairleadsCalc2", MDdll), MDFairCalcParams)
MDConn = MDConnProto(("GetConnectPos", MDdll), MDConnParams)
MDFair = MDFairProto(("GetConnectForce", MDdll), MDFairParams)
MDClose= MDClosProto(("LinesClose", MDdll))

# set up passable input and output variables/arrays  
x1 = (c_double*6)()                   # coupling point 6DOF position/rotation vector (if using MDCalc)
xd1 = (c_double*6)()
f = (c_double*6)()                   
x = (c_double*(3*nFair))()  # coupling points 3DOF position vectors (if using  MDFairCalc)  [x1,y1,z1, x2,y2,z2...]
xd = (c_double*(3*nFair))()
ff = (c_double*(3*nFair))()
t = (c_double*1)()                # pointer to t
dt= (c_double*1)()                # pointer to dt

# time steps setup
dtC = 0.02                  # coupling time step size
ts = np.arange(0,300,dtC)  
t[0] = 0
dt[0] = dtC

# initial conditions
for i in range(6):
        x1[i]=0.0
        xd1[i] = 0.0

# initialize MoorDyn at origin
MDInit(x1,xd1)

xold = np.zeros(3*nFair)

# do time steps
for i in range(20): #len(ts)):
        t[0] = ts[i]
        dt[0] = dtC
        
    # set vessel node motions to be whatever you like
        x[0] = ...
        x[1] = ...
        x[2] = ...
        x[3] = ...
        x[4] = ...
        x[5] = ...
               
        # calculate velocities with finite difference
        if i==0:
               for j in range(nFair):
                       for k in range(3):
                              xd[3*j+k] = 0.0
                               xold[3*j+k] = 0.0
        else:
               for j in range(nFair):
                       for k in range(3):
                              xd  [3*j+k] = (x[3*j+k] - xold[3*j+k])/dtC
                              xold[3*j+k] = x[3*j+k]

        # call the MoorDyn time step function                       
        result = MDFairCalc(x, xd, t, dt)  # this is the one for coupling about individual fairlead points
        
# free memory when done
MDClose()      
```

yonghoonlee, 2021-06-16:
@trashid10 and @mattEhall 
Thank you for the guidance. Especially the code snippet that @mattEhall helped a lot making MoorDyn working on my python code.

closed 2021-06-16 by yonghoonlee