FloatingArrayDesign/MoorDyn #7: Can MoorDyn resolve the vessel node's position by itself?

trashid10, 2021-06-15:
I am trying to resolve a simple case where a ball is attached to a line of some length and being pulled (basically, a ball being towed). I have my lines.txt file specified such that:

- My line has coefficients
- Node 1 is fixed at 0,0,-5
- Node 2 is a vessel at 0,0,-60 with a mass that is ~50% more than its buoyant equivalent along with a drag coefficient and volume
- I define that node1 is anchor and node 2 is fair, and I specify an unstrlen of 55

I am using the "FairleadsCalc2" function and inputting position, velocity, flines, time, and dt, e.g:
```
for i in range (1,time_steps+1):
     MoorDyn.FairleadsCalc2(position, velocity, flines_p, time_array[i-1], ctypes.c_double(dt))
```

If my position array were [0 0 0 0 0 0], then the vessel's position moves to that location. Similarly, if my surge velocity were 1, the vessel moves to 1*time_step. What I would like to do is have the vessel node to move on its own (based on its mass and drag term, given some input velocity)...but is that possible with MoorDyn "out of the box"? 

Pic related is what I would like to resolve:
[image]

Perhaps I am mixing something up (or using the wrong function), but all hints are appreciated!

mattEhall, 2021-06-16:
Hi - I'm not fully understanding what you want to do. 
When you say
> "What I would like to do is have the vessel node to move on its own (based on its mass and drag term, given some input velocity)"

do you mean "given some _initial_ velocity"? 

It is possible to have a fully free-floating system in MoorDyn, although the model probably doesn't have the capabilities you'd need for accurately modeling something on the water surface. If you clarify what you're looking for we can probably find a solution.

trashid10, 2021-06-16:
Taking a step back - I want to model the picture above (a ball being towed), with MoorDyn. What I did is make two nodes, one anchor and one vessel (the vessel node being the ball). The vessel node is given a mass, volume, and some drag coefficient. Now, I want to take my Mooring system and solve for the position of the ball after a long time. Right now, driving MoorDyn with Python forces me to specify the position and velocity of the vessel node (aka the ball)...but is there a way for me _not_ to specify the position and velocity and let MoorDyn handle the motion?

mattEhall, 2021-06-19:
Ah, I think I see what you're saying. When you mentioned [0,0,0,0,0,0] and the vessel moving to that location I thought you were referring to the ship in your image. You shouldn't have any problem doing what you want to do -- just a few things need to be straightened out.

The node point that attaches to the ship in your image should be "vessel" type, because that's what you want to prescribe the motion of (using size-3 position and velocity inputs).

The node point that is the suspended ball should be "connect" type, as you suspected, because you want MoorDyn to solve its dynamics.

Connect the ends of the line to these nodes (it doesn't matter which end is which) and you should be good to go!
If there are still some problems, share your input file and we can get to the bottom of it!

trashid10, 2021-06-21:
Thank you, your suggestion worked and the results seem reasonable. 

I will comment that "Cda" for the connect node does not seem to do anything (I can make Cda = anything and my line's x forces don't change). Unless I misunderstood something, the Cda I am using is the drag coefficient multiplied by projected area. The x tension therefore is off ~equal to the node's missing resistance. Even if I give the node an "FX" equal to the missing resistance, cable's resulting x forces don't change. Perhaps there is an accounting error or I maybe I am doing something wrong.

closed 2023-03-24 by sanguinariojoe