FloatingArrayDesign/MoorDyn #9: MATLAB API

TGiles1998, 2022-01-05:
Hi,

I am really struggling to get MATLAB to talk to moordyn, is there nay chance a document with more in detail set up process and API can be added?  

Many thanks

TGiles1998, 2022-01-05:
Possibly an example MATLAB script?

Thanks again

TGiles1998, 2022-01-06:
I first tried to compile the software on mac and it produced the errors:

make
clang++ -c -Ofast -g -w -Wall -static -static-libgcc -static-libstdc++ -std=gnu++0x -DMoorDyn_EXPORTS -DOSX ../MoorDyn.cpp
In file included from ../MoorDyn.cpp:20:
../MoorDyn.h:41:5: error: unknown type name 'dllexport'
int DECLDIR LinesInit(double X[], double XD[]);
    ^
../MoorDyn.h:22:30: note: expanded from macro 'DECLDIR'
                #define DECLDIR __declspec(dllexport)
                                           ^
../MoorDyn.h:41:13: error: expected function body after function declarator
int DECLDIR LinesInit(double X[], double XD[]);
            ^
../MoorDyn.h:43:5: error: unknown type name 'dllexport'
int DECLDIR LinesCalc(double X[], double XD[], double Flines[], double* , double* );
    ^
../MoorDyn.h:22:30: note: expanded from macro 'DECLDIR'
                #define DECLDIR __declspec(dllexport)
                                           ^
../MoorDyn.h:43:13: error: expected function body after function declarator
int DECLDIR LinesCalc(double X[], double XD[], double Flines[], double* , double* );
            ^
../MoorDyn.h:45:5: error: unknown type name 'dllexport'
int DECLDIR FairleadsCalc2(double rFairIn[], double rdFairIn[], double fFairIn[], double* t_in, double *dt_in); // easier to call version

I then switched over to windows operating system using the already pre-compiled .dll file but I could not get MATLAB to load in the library. The API I tried to use is from the manual:  http://www.matt-hall.ca/files/MoorDyn%20Users%20Guide%202017-08-16.pdf

%% Setup
X = zeros(6,1); % platform position
XD = zeros(6,1); % platform velocity
N = 10; % number of coupling time steps
dt = 0.5; % coupling time step size (time between MoorDyn calls)
Ts = zeros(N,1); % time step array
FairTens1 = zeros(N+1,1); % array for storing fairlead 1 tension time series
FLines_temp = zeros(1,6); % going to make a pointer so LinesCalc can modify FLines
FLines_p = libpointer('doublePtr',FLines_temp); % access returned value with FLines_p.value
%% Initialization
loadlibrary('Lines','MoorDyn'); % load MoorDyn DLL
calllib('Lines','LinesInit',X,XD) % initialize MoorDyn

%% Simulation
XD(1) = 0.1; % give platform 0.1 m/s velocity in surge
for i=1:N
 calllib('Lines', 'LinesCalc', X, XD, FLines_p, Ts(i), dt); % some MoorDyn time stepping
 FairTens1(i+1) = calllib('Lines','GetFairTen',1); % store fairlead 1 tension
 X = X + XD*dt; % update position
 Ts(i+1) = dt*i; % store time
end

%% Ending
calllib('Lines','LinesClose'); % close MoorDyn
unloadlibrary Lines; % unload library (never forget to do this!)

The error I get for MATLAB is: 

Error using loadlibrary
Failed to preprocess the input file.
 Output from preprocessor is:cc1.exe: fatal error:
 C: ...\MoorDyn: No such file or directory
compilation terminated.

Error in MoorDyn_API (line 11)
loadlibrary('Lines','MoorDyn'); % load MoorDyn DLL

TGiles1998, 2022-01-06:
Partially Solved:

loadlibrary('Lines','MoorDyn'); % load MoorDyn DLL

has been changed to:

loadlibrary('Lines','MoorDyn.h'); % load MoorDyn DLL

The compiled library from the WEC-SIM has been used to work with MAC OS: https://github.com/WEC-Sim/MoorDyn

The software will now load into MATLAB but issues arising for the two lines:

X = X + XDdt; % update position
Ts(i+1) = dti; % store time

daankoetzier, 2022-02-28:
Hi @TGiles1998

I am trying to access MoorDyn through MATLAB in Windows in a similar way, and I am receiving similar error messages about libraries not being loaded. I looked at your proposed solutions, but I still cannot get the software to work due to missing libraries. I think the problem is in the file setup in my working folder, could you please share the files you are using and the folder structure?

Many thanks for your help!

TGiles1998, 2022-02-28:
Hi @daankoetzier 

For windows I also then changed 

loadlibrary('Lines','MoorDyn.h'); % load MoorDyn DLL

to 

loadlibrary('Lines.dll','MoorDyn.h'); % load MoorDyn DLL

adding the .dll

My folder structure is as follows:

-Overall Folder:
 - Lines.dll
 - Matlab_API_commands.m
 - mooring (another folder)
   - lines.txt

Hope this helps!

TGiles1998, 2022-03-01:
Hi, You have to either compile the software yourself from the make file in the repo or alternatively you can access it in its already compiled version that is part of the WEC-SIM. Link here: https://github.com/WEC-Sim/moorDyn

Download these from the WEC-SIM repo using the link and setup the folder as previously mentioned and it should work.

Hope this helps!

TGiles1998, 2022-03-01:
I've opened a pull request so that these files can be more easily found from the main repo. https://github.com/mattEhall/MoorDyn/pull/17#issue-1155927336
Thanks,
Thomas

TGiles1998, 2022-03-02:
@hwen42 

Yes just change the line.txt file with your modified one. The compiled dynamic library will just pick this up and run it through the C++.

hwen42, 2022-03-02:
@TGiles1998 Thanks for answering. I downloaded the WEC_Sim-master folder and tried to run the example. It works. But the only lines.txt file that I found is in RM3_CoupledWithMoorDyn folder. If I have to replace with my own input txt file. Should I just replace it in that directory? Or should I make a new morring folder and put it there,

hwen42, 2022-03-02:
This is the Input file that I have:
Mooring line data file for MoorDyn in Lines.dll
---------------------- LINE DICTIONARY -----------------------------------------------------
LineType  Diam    MassDenInAir    EA          BA/-zeta     Can     Cat    Cdn     Cdt 
(-)       (m)       (kg/m)        (N)         (Pa-s/-)     (-)     (-)    (-)     (-) 
chain     8e-3      1.1           7.5E+6          0         0      0      2.4     1.15
---------------------- NODE PROPERTIES -----------------------------------------------------
Node      Type      X        Y         Z        M        V        FX       FY      FZ     CdA   CA
(-)       (-)      (m)      (m)       (m)      (kg)     (m^3)    (kN)     (kN)    (kN)   (m^2)  (-)
1        Fix         0       0         0        0        0        0        0       0       0     0
2        Vessel      5       0     -8.6603      0        0        0        0       0       0     0

---------------------- LINE PROPERTIES -----------------------------------------------------
Line     LineType  UnstrLen  NumSegs   NodeAnch  NodeFair  Flags/Outputs
(-)      (-)       (m)         (-)       (-)       (-)       (-)
1        chain     10        5           1        2        tp
---------------------- SOLVER OPTIONS-----------------------------------------
0.001    dtM          - time step to use in mooring integration
0        WaveKin      - wave kinematics flag (0=neglect, the only option currently supported)
70       WtrDpth      - water depth
5.0      CdScaleIC    - factor by which to scale drag coefficients during dynamic relaxation IC gen
0.001    threshIC     - threshold for IC con
-------------------------- OUTPUTS --------------------------------
FairTen1
Con1px Con1py Con1pz Con2px Con2py Con2pz
--------------------- need this line ------------------

hwen42, 2022-03-02:
> 

Hello Matt, there is actually no line above the "Line Properties" (It just looks like that when I copy pasted.) I am just trying to simulate a simple problem where one end is fixed and one end is free to move. Therefore I only need 2 node. But when I used the example Matlab code. I can't generate any results while I can generate result using the example provided (the one with 6 nodes). My question is do I have to modify the dll file in the functions/moordyn folder?

sanguinariojoe, 2022-08-10:
The [v2 candidate](https://github.com/mattEhall/MoorDyn/pull/31) has a Matlab wrapper, based on mex files, which is blazzing. Unfortunately, it is totally impossible to build that wrapper automatically in a github action, so no binaries will be provided. That means you have to compile MoorDyn by yourself to get Matllab support. The process is documented [here](https://github.com/sanguinariojoe/MoorDyn/blob/v2/docs/starting.rst#compile-and-install-moordyn)

An example of a Matlab script can be found [here](https://github.com/sanguinariojoe/MoorDyn/blob/v2/docs/coupling.rst#matlab)

sanguinariojoe, 2023-01-07:
Can we close this?