FloatingArrayDesign/MoorDyn #13: Compile Moordyn And use it with Matlab

salicto, 2022-02-02:
Hello,

I would like to use moorDyn and Map++ separately (don't include it in the OpenFAST code). I would like to plotted the anchor lines. However I can't generate the DLL file. Usually I use visual studio 2017 and the nmake command. Could you share the process to be able to use these codes with matlab or python?

Best Regards,

mattEhall, 2022-02-08:
Hi, in Windows I usually use MinGW and its "minGW32-make", with the makefiles that are included in the MoorDyn repo. Interacting with the DLL from Python or Matlab is the same as interacting with any other DLL from those languages. Let me check if there's an example somewhere yet...

sanguinariojoe, 2022-08-10:
The [v2 candidate](https://github.com/mattEhall/MoorDyn/pull/31) has a Matlab wrapper, based on mex files, which is blazzing. Unfortunately, it is totally impossible to build that wrapper automatically in a github action, so no binaries will be provided. That means you have to compile MoorDyn by yourself to get Matllab support. The process is documented [here](https://github.com/sanguinariojoe/MoorDyn/blob/v2/docs/starting.rst#compile-and-install-moordyn)

An example of a Matlab script can be found [here](https://github.com/sanguinariojoe/MoorDyn/blob/v2/docs/coupling.rst#matlab)

Echo-new, 2022-11-24:
Hi，I have a problem about v2 candidate.When I configure moordyn,it shows that the source directory "C:/MoorDyn/MoorDyn" does not appear to contain CMakeLists.txt.So,I want to know where to find the CMakeLists.txt.

sanguinariojoe, 2022-11-24:
CMakeLists.txt is in the root folder of MoorDyn. I suspect you have not cloned/downloaded the v2 candidate:

https://github.com/mattEhall/MoorDyn/tree/v2

But the v1:

https://github.com/mattEhall/MoorDyn

Can you check that? (They have radically different folders and files)

Echo-new, 2022-11-25:
I do used the v1 files, thank you so much. 
And now I have a new question, when I generate moordyn, it shows that  target "moordyn" requires the language dialect "CXX17" (with compiler extensions).  But the current compiler "GNU" does not support this, or CMake does not know the flags to enable it. 
Maybe you know how to solve it?

sanguinariojoe, 2022-11-25:
MinGW absolutely supports -std=c++17. Can you tell me which version do you have? How did you install it?

Can you also paste here the exact output of cmake?

Echo-new, 2022-11-26:
I reinstalled the MinGW (from githup ,version :x86_64-12.2.0-release-win32-seh-rt_v10-rev1.7z)and added the path. 
Then it worked, thank you again. But when I use a cmd with administrative rights , it produces errors. How should I solve this?
[image]
[image]

sanguinariojoe, 2022-11-27:
First, do not crop the output, it makes way harder to analyze the problem. On top of that, text is better than image.

MoorDyn is compiling just fine. Apparently it is failing while building the Python wrapper. So I suggest you to just disable it (PYTHON_WRAPPER=OFF in CMake)

Echo-new, 2022-11-28:
I got it, thank you very much.

sanguinariojoe, 2023-01-07:
Can we close this?

xdd1997, 2023-02-11:
Dear @sanguinariojoe ：

I have some problems in the process of calling Moodyn with Matlab on windows 10, Your help is greatly appreciated!

## problem description

I'm compiling eigen by CMake and I'm getting the following error, 

> CMake Error: CMake was unable to find a build program corresponding to "MinGW Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
> CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
> CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
> Configuring incomplete, errors occurred!

The computer's environment variables has been have included:

> E:\msys64
> E:\mingw64
> E:\mingw64\bin

## attachment

[image]

xdd1997, 2023-02-11:
@sanguinariojoe 

 Hi, I want to call the Moordyn program on windows using matlab. I was following this tutorial $[1]$,  and there was a problem compiling the MoorDyn input `mingw32-make` in cmd as Administrator. The output was:

mingw32-make[2]: *** [wrappers\matlab\CMakeFiles\MoorDynM_Close.dir\build.make:76: wrappers/matlab/CMakeFiles/MoorDynM_Close.dir/MoorDynM_Close.cpp.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:1089: wrappers/matlab/CMakeFiles/MoorDynM_Close.dir/all] Error 2
mingw32-make: *** [Makefile:165: all] Error 2

Thanks very much for your help.

[1] https://github.com/sanguinariojoe/MoorDyn/blob/v2/docs/starting.rst#compile-and-install-moordyn

sanguinariojoe, 2023-02-17:
Hey @xdd1997 !

This is not a problem on MoorDyn, but on Matlab (or eventually on MinGW). I will anyway give it a shot as soon as I can find a windows machine with Matlab

sanguinariojoe, 2023-02-17:
@xdd1997 I uploaded a couple of fixes to https://github.com/sanguinariojoe/MoorDyn/tree/v2, but the compilation is working to me with the steps discussed above

xdd1997, 2023-02-18:
**Thank you for your help**. I have successfully compiled many `*.mexw64` files, and copied these `*.mexw64` files and `libmoordyn.dll` file to the project path. The demo can run smoothly now. 

I suggest you to modify this part of the tutorial<sup>[1]</sup> as my error started from this point:

[image]

I was unable to obtain the `mimgw32-make.exe` program with the above command, so I obtained the mingw64 program through the link<sup>[2]</sup>  However, the error I mentioned earlier  occurred.

I finally managed to compile it by typing `pacman -S mingw-w64-x86_64-toolchain` in MSYS2 MINGW64.

Anyway, thank you again for your help.

[1]  https://github.com/sanguinariojoe/MoorDyn/blob/v2/docs/starting.rst#compile-and-install-moordyn
[2]  https://www.mingw-w64.org/downloads/#mingw-builds.