FloatingArrayDesign/MoorDyn #21: Questions regarding Running Input File in Matlab

hwen42, 2022-03-04:
Hello,

I am new to Moordyn and I would like to model a simple problem where I want to fix one chain at one end while letting the other end moves freely. I am wondering do I still need two nodes to specify where are the ends attached to? Or can I just use one node for the fixed end and not specify the free end. Thank you!

Haiqi Wen

sanguinariojoe, 2022-08-10:
The [v2 candidate](https://github.com/mattEhall/MoorDyn/pull/31) has a Matlab wrapper, based on mex files, which is blazzing. Unfortunately, it is totally impossible to build that wrapper automatically in a github action, so no binaries will be provided. That means you have to compile MoorDyn by yourself to get Matllab support. The process is documented [here](https://github.com/sanguinariojoe/MoorDyn/blob/v2/docs/starting.rst#compile-and-install-moordyn)

An example of a Matlab script can be found [here](https://github.com/sanguinariojoe/MoorDyn/blob/v2/docs/coupling.rst#matlab)

sanguinariojoe, 2023-01-07:
Can we close this?

closed 2023-03-24 by sanguinariojoe