FloatingArrayDesign/MoorDyn #28: Program fails to create Mooring system with large numbers of output files

djanders19, 2022-07-06:
With large numbers of output files (1022+ on my system) MoorDyn throws an error when creating .out files:
`Cannot create the output file './moordyn_inputs/test_with_10000_lines_Line1022.out
Error (-2) during the Mooring System creation:
Exception while reading the input file`

I believe this to be an issue with how files are handled. After outfiles are created, they appear to be kept open until the MoorDyn deconstructor is called. Most systems have a limit on the maximum number of open files at any given time, so I believe closing output files when they are not being actively written to or read from should solve the problem. I have tested closing the output files after they are created and this appears to solve the above error - however, it introduces more bugs as the downstream methods assume that all the outfiles are open.

There may not be many use-cases where this is an issue (I don't know how large people's models tend to get) so this refactor may not be worth the time/energy and presumed performance hit from opening and closing every output file every time the program wishes to write outputs. A more extensive, and possibly better, refactor would be to have all output saved to one larger file (presumably following Tidyverse principles or something similar) and just letting the user pull out individual Object outputs as part of their processing pipeline.

mattEhall, 2022-07-07:
This is a good point of discussion. I have rarely heard of scenarios that require more than 100 or 200 Lines -- even with some of our shared mooring array simulations, but I can imagine there will be important exceptions. For how to handle it, my thinking is to have a single file as you suggest. Along with that, have been hoping to add the capability for binary output files to save space and time. We should keep this in mind and discuss more as we get closer to finishing version 2!

sanguinariojoe, 2022-08-10:
The [v2 candidate](https://github.com/mattEhall/MoorDyn/pull/31) is already able to save binary files, which are BTW portable and platform agnostic.

It is though dangerous to save them in a double precission compiled MoorDyn (the standard) and load them in a single precission one (which is an option to boost the performance at the cost of accuracy).

I am planning to add VTK support for post-processing, but that will happens after  [the PR](https://github.com/mattEhall/MoorDyn/pull/31) is merged

djanders19, 2022-08-10:
So does the binary file update resolve the limited-line-number issue? I can see how it could reduce memory requirements quite a bit, but my understanding was that with large numbers of lines the program actually runs into an OS imposed limit on the number of files that can be open (as opposed to simply running out of memory). Could be totally wrong about that though - when I get some time I can try cloning that PR and running some 600-1000 line tests through to see if it still crashes the program.

sanguinariojoe, 2022-08-10:
That binary file is a single standalone file, so no OS limit will be reached. Said that, such file is not meant for post-processing tasks at all, but to can resume the simulation. That's why I suggest implementing VTK.

We can also look for some other text-based legacy format to replace the current files .out files. [.dxf](https://github.com/codelibs/libdxfrw) maybe?

sanguinariojoe, 2022-10-06:
In my opinion all the input/output shall be overhauled, giving the user more control through the API. On the other hand, I suppose @mattEhall would always keep compatibility with FAST, so the input and output files would be always there.

But that is not happening in the v2.0 version. So your way seems nice to me. Let's see what @mattEhall says, because that would affect the way FAST is working.

I will push him to merge [the big PR](https://github.com/mattEhall/MoorDyn/pull/31) so we do not have much conflicts. It is anyway merging in a v2 branch, so it should not be a problem

sanguinariojoe, 2022-10-06:
P.S. I already added VTK support. Checkout [the big PR](https://github.com/mattEhall/MoorDyn/pull/31)

djanders19, 2022-10-06:
I hear you on the IO front - I do think we'll need to preserve the old output formats so as to not break user post-processing. I don't have any great ideas for reducing that coupling, though developing an I/O portion of the API could be a good step in the right direction as you suggest.

Right now I'm just letting the user specify whether to output to many or one file as a parameter in the input file - so by default every Line has its own output, but it's possible to tell the program to dump them all in one file. I've also been limiting the feature to just Lines so far, as that's what we need for our work, but should be pretty easy to expand to Rods and Bodies in the new OO framework. Gonna go work on that right now in fact.

re: VTK, sounds great! I'm not too familiar with VTK (I've mostly used OpenGL for past vis work), so this may be a naive question, but is there a way to extract numerical data from the VTK outputs (or the binary file in your candidate)? A lot of our post-processing work has little to do with visualizing the actual system, but requires the raw time series of tensions, positions etc.

sanguinariojoe, 2022-10-06:
> I've also been limiting the feature to just Lines so far, as that's what we need for our work, but should be pretty easy to expand to Rods and Bodies in the new OO framework

My VTK implementation is also limited to lines now. But I am planning to extend that to every single object in the following weeks.

> so this may be a naive question, but is there a way to extract numerical data from the VTK outputs (or the binary file in your candidate)?

Sure! As you can see [here](https://github.com/mattEhall/MoorDyn/pull/31/commits/e8fb2b4c97b0eb85aeaf32ab5f4ca34a423104f8#diff-29898bc00510b247ddb019e5cb3b3c646b35ae51e549647eebcf010eb1899fa6R1413) I am packing the numerical information of both points/nodes and segments. That can be read afterwards in your post-processor. [In Python it is particularly convenient](https://www.programcreek.com/python/example/108192/vtk.util.numpy_support.vtk_to_numpy).

sanguinariojoe, 2023-01-07:
I think we can close this

sanguinariojoe, 2023-06-20:
@djanders19 ? Any feedback? Can we close this issue? What about the PRs?

https://github.com/FloatingArrayDesign/MoorDyn/pull/35
https://github.com/FloatingArrayDesign/MoorDyn/pull/37

Are they outdated?

closed 2023-06-21 by sanguinariojoe