FloatingArrayDesign/MoorDyn #30: Empty output file Lines.out on Linux OS

AB1982, 2022-07-19:
Dear Matt

I am using the Moordyn library with python on a Linux OS system.
Everything work fine except the output file. 
The library creates the Lines.out file, but it s almost empty... it appears as follow:

``
Time``
``	 
(s)
``
only the header for the time and its unit.
Do you have any suggestions? Maybe I have to change some lines in the source code? 
Thanks

djanders19, 2022-07-20:
This is odd - I've had no trouble running MoorDyn on a Linux OS (save trying to run it on a server where the compiler was hopelessly out of date and I couldn't get root access... 😞), though admittedly I've only called it from cpp programs.

Would you be willing to share your input file and .log file? My initial guess would be that there could be an error within the input file, though from the limited description above it sounds like the program is not running at all or is somehow getting hung up during dynamic relaxation/the initial time-step... I believe these headers are written to the outfiles immediately after the input is read in. Are you certain that MoorDyn is running to completion before your calling script returns?

AB1982, 2022-07-20:
Hi
here attached the input file
[lines.txt](https://github.com/mattEhall/MoorDyn/files/9154536/lines.txt)
The same input file, tested on a Windows OS, works fine and in the output file (Lines.out) I can read the velocities, positions and tensions for each nodes. 
The results we get (the tension on the fairlead), during the execution of the python script  on both the OS, are the same... 
Concerning the log file I cannot see it...The only message I get from moordyn is the header during its initialization:
``
 Running MoorDyn (v1.01.03C, 2021-03-09)
   Copyright (c) Matt Hall and NREL, licensed under GPL v3.
   Creating mooring system.  
``

djanders19, 2022-07-21:
Got it - I see you're running v1, which I'm not as familiar with. I've mainly worked with v2 (currently on the dev branch). Just to get the (maybe) obvious issues out of the way: are you linking .SO libraries on the Linux script, or still trying to link in .dlls? .dlls generally don't work on Linux machines unless they've been compiled with the .NET core.

If you can provide some more info about your environment (OS, compiler, etc.) then I can try to replicate the issue this weekend.

AB1982, 2022-07-21:
Obviously, on Linux OS (Debian 11) I use the .so lib, compiled by me by the makefile provided in the compileSO folder. The compiler used is g++, here the output of ``g++ -v`` command:
``
gcc version 10.2.1 20210110 (Debian 10.2.1-6)``
On Windows the student use the precompiled libs (.dll).

sanguinariojoe, 2022-08-10:
v2 will have its own Python wrapper, see [the PR](https://github.com/mattEhall/MoorDyn/pull/31). Indeed a pip packet is already available on TestPypi. You can test it typing:

`pip install --index-url https://test.pypi.org/simple/ moordyn`

Do you have a minimal sample script [here](https://github.com/sanguinariojoe/MoorDyn/blob/v2/docs/coupling.rst#python)

P.S. You are welcome to support this v2 candidate in the same [PR link](https://github.com/mattEhall/MoorDyn/pull/31):

[image]

So it will be merged ASAP. You can also leave your feedback

closed 2023-03-24 by sanguinariojoe