FloatingArrayDesign/MoorDyn #32: Tension when using Connect with Moredine in OpenFAST

yk202208, 2022-08-04:
Why is the line tension other than the connecting part of the floating body smaller by several orders of magnitude when using Connect in OpenFAST?

I've seen this issue at least with OpenFAST2.4 and OpenFAST3.0.
For example, when connecting with anchor-chain①-rope-chain②-vessel, the tension of chain① and rope becomes extremely low compared to the tension of chain②. I think the chain② tension is at a reasonable level, but I would like to know the correct rope and chain① tension as well.

I can't share all the information because it is related to the confidentiality of the project, but if you connect like the example, FairTen 7-9 will output a reasonable tension, but FairTen 1-6 will output a value that is several orders of magnitude smaller.

yk202208, 2022-08-04:
Sorry, I forgot to attach the sample.

[sample_MoorDyn.txt](https://github.com/mattEhall/MoorDyn/files/9255811/sample_MoorDyn.txt)

mattEhall, 2022-08-06:
See 432349
This will be changed in the future to match the behavior you're expecting.

sanguinariojoe, 2023-03-24:
Is this fixed then? Can we close this?

closed 2023-09-12 by sanguinariojoe