FloatingArrayDesign/MoorDyn #40: OpenFOAM coupling

sanguinariojoe, 2022-12-22:
Just to let you know I upgraded the coupling with OpenFOAM to MoorDyn v2:

https://gitlab.com/hfchen20/foamMooring/-/merge_requests/1

hfchen20, 2023-01-18:
Running v2 in WSL:

Attached is the v2 input file, in which fairlead tensions such as FairTen1 and FairTen3 are requested.

- In 'lines_v2.out', only time was output, but not tensions as requested in OUTPUTS section of the input file.
- The time values in all the output files are incorrect if moordyn_step() is called more than once per time step. Several iterations are normally needed to ensure coupling between CFD solver and body motion. The time log (deltaT, to be specific) was likely added as many times as the number of iterations per time step.

[lines_v2.txt](https://github.com/mattEhall/MoorDyn/files/10451150/lines_v2.txt)

[image]

sanguinariojoe, 2023-01-19:
> In 'lines_v2.out', only time was output, but not tensions as requested in OUTPUTS section of the input file.

I will take a look on this right now

> The time values in all the output files are incorrect if moordyn_step() is called more than once per time step. Several iterations are normally needed to ensure coupling between CFD solver and body [...]

I have bad news for you... If this is the case then your V1 version is wrong too. When you call to `LinesCalc()` (no matter what parameters do you use, provided that the time step is not null), the stack you are getting is:

```
...
rk2()
FairleadsCalc()
LinesCalc()
```

`rk2()` is called with the global variable states, which is always updated forward (see https://github.com/mattEhall/MoorDyn/blob/master/MoorDyn.cpp#L139).

In other words, if you call `n` times `LinesCalc()` within a time step, with a time step `dt`, then your lines are actually behaving like you are calling to `LinesCalc(..., n * dt)` once. It is not exactly like that, because some weird stuff will be happening with the fairleads, but it will be so far similar.

I think I can fix this for v2, but v1 is surely doomed...

hfchen20, 2023-01-19:
For v1 output (summary or each line), I did see n entries for n iterations per time step. n=3 for snapshot below.

[image]

sanguinariojoe, 2023-01-20:
Which is indeed wrong. The absolute time is only involved in some interpolations, not in the actual computations. Each time you see a timestamp (no matter if it is repeated), the dynamics of your lines have been integrated forward 1 whole time step. Thus, your coupling is only working approximately fine for quasi-static cases.

I cannot check Moody because it is not open source, but I suspect it has similar issues (and if it does not, it has problems integrating long time periods). Map++ is quasi-static, so you are safe

closed 2023-03-24 by sanguinariojoe