FloatingArrayDesign/MoorDyn #43: MoorDyn with MATLAB

relhamoud, 2023-01-14:
Dear @mattEhall ,

I am trying to couple MoorDyn in Matlab. I used the script in MoorDyn user's guide.
However i am getting the following error.

Could you please help me ?

[image]

sanguinariojoe, 2023-01-25:
Have you tried [MoorDyn v2](https://github.com/mattEhall/MoorDyn/tree/v2)? You have the documentation [here](https://moordyn-v2.readthedocs.io/en/latest/) and even [a matlab test](https://github.com/mattEhall/MoorDyn/blob/v2/tests/matlab_bindings.m.in)

hfchen20, 2023-02-09:
@sanguinariojoe Could you please help me figure out what went wrong? Thank you.

I compiled v2 with fortran wrapper on, and tried to link moordyn (moordynf) with a simple c (fortran) project, as demonstrated at https://moordyn-v2.readthedocs.io/en/latest/starting.html#c. Both c and fortran tests resulted in link error, which points to the component.
Imported target "MoorDyn::moordyn" includes non-existent path, ...
Imported target "MoorDyn::moordynf" includes non-existent path.

    "$FOAM_USER_LIBBIN/COMPONENT"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.
  * An install or uninstall procedure did not complete successfully.
  * The installation package was faulty and references files it does not
  provide."

Below is what I found in $FOAM_USER_LIBBIN/cmake/moordyn/MoorDynTargets.cmake 
- target MoorDyn::moordyn 
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;${_IMPORT_PREFIX}/COMPONENT;${_IMPORT_PREFIX}/Runtime" 
- target MoorDyn::moordynf 
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include"

sanguinariojoe, 2023-02-09:
Hey @hfchen20 !

This has nothing to do with Matlab, right? Would you please open a new issue? And in that issue please let me know the list of commands you have typed to compile and install MoorDyn

hfchen20, 2023-02-09:
@sanguinariojoe Nothing to do with matlab. Opened new issue #47