FloatingArrayDesign/MoorDyn #47: Compile v2 and link error with own project

hfchen20, 2023-02-09:
I compiled v2 with fortran wrapper on, and tried to link moordyn (moordynf) with a simple c (fortran) project, as demonstrated at https://moordyn-v2.readthedocs.io/en/latest/starting.html#c. Both c and fortran tests resulted in link error, which points to the non-existing component.
Imported target "MoorDyn::moordyn" includes non-existent path, ...
Imported target "MoorDyn::moordynf" includes non-existent path.

"$FOAM_USER_LIBBIN/../COMPONENT"
in its INTERFACE_INCLUDE_DIRECTORIES. Possible reasons include:

The path was deleted, renamed, or moved to another location.
An install or uninstall procedure did not complete successfully.
The installation package was faulty and references files it does not
provide."

Below is what I found in $FOAM_USER_LIBBIN/cmake/moordyn/MoorDynTargets.cmake

- target MoorDyn::moordyn (the moordynf has the same entry)
INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;${_IMPORT_PREFIX}/COMPONENT;${_IMPORT_PREFIX}/Runtime"

The v2 location is a module folder as in https://gitlab.com/hfchen20/foamMooring/-/blob/master/Allwmake. The compiling command is as follows:
> mkdir MoorDyn.build
cd MoorDyn.build 
cmake -DBUILD_TESTING=ON -DEXTERNAL_EIGEN=OFF -DFORTRAN_WRAPPER=ON -DMATLAB_WRAPPER=OFF -DPYTHON_WRAPPER=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$FOAM_USER_LIBBIN/.. ../MoorDyn 
make -j 
make install 

As a side note, there was CMake error (Eigen3Config not found) when linking with own project, even I compiled with EXTERNAL_EIGEN=OFF. Is this due to the potential conflict `find_dependency(Eigen3 3.3)` in https://github.com/mattEhall/MoorDyn/blob/v2/cmake/MoorDynConfig.cmake? I had to comment off this link to expose the target linking error above.

sanguinariojoe, 2023-02-12:
Hey @hfchen20 , here you have a solution. Please, notice that there you do not have the solution to #40  (which is in this PR: https://github.com/mattEhall/MoorDyn/pull/44)

I will probably merge all the bugfixes on my fork v2 branch (https://github.com/sanguinariojoe/MoorDyn/tree/v2) and ping you back

hfchen20, 2023-02-12:
> I will probably merge all the bugfixes on my fork v2 branch

@sanguinariojoe It would be very nice to have all those features in one place. Thank you.

sanguinariojoe, 2023-02-14:
Hey @hfchen20 !

I merged all the PRs in one big monster:

https://github.com/mattEhall/MoorDyn/pull/51

It is building and testing right now: https://github.com/sanguinariojoe/MoorDyn/actions/runs/4174069439

closed 2023-03-24 by sanguinariojoe