FloatingArrayDesign/MoorDyn #54: Compile issue with setup.py

RyanDavies19, 2023-03-08:
When compiling using clang on MacOS, setup.py gives errors because vtkSmartPointer.h, vtkPolyData.h, vtkCharArray.h, vtkFloatArray.h, and vtkErrorCode.h are not found. Are this files that can be added to the repo?

sanguinariojoe, 2023-03-09:
Hey hey!

Until the repo is moved to its new home, you need to summon me writing @sanguinariojoe (otherwise I will never get notified)

The problem here is that setup.py is not meant for the users. Look at this comment: https://github.com/mattEhall/MoorDyn/blob/v2/setup.py#L34

A bit more detailed explanation...

Users can install the automatically built binaries (https://github.com/mattEhall/MoorDyn/releases/tag/nightly), or they can compile MoorDyn themselves using CMake (see https://moordyn-v2.readthedocs.io/en/latest/starting.html#compile-and-install-moordyn).

CMake can build the C/C++ library, as well as the Matlab, Python and Rust wrappers. User can also tweak some other options. At some point we should even drop the legacy Makefiles.

So Cmake can cover everything... Except the auto-built Python packages for Pypi... They are built with https://github.com/pypa/cibuildwheel, which requires a setup.py (well, nowadays the appropriate way to do this is by a project.toml file).

Regarding the errors you have noticed, they are happening because those packages are built with VTK support, using this other tool I created:

https://github.com/sanguinariojoe/vtk-builds

It is very little tested though...

The solution to this issue is not making this setup.py more versatile, but to move it somewhere that users might not find it. I'll do that!

RyanDavies19, 2023-03-09:
I think it would be best to leave setup.py where it is for now. I have been using it so that I can run MoorDyn with local changes using both the new and old API's. Is there any simple way to work around this for now if I want to use setup.py locally?

sanguinariojoe, 2023-03-09:
The v2 absolutely! CMake can perfectly take care of that. Trying to maintain a versatile enough setup.py will make our job harder

If by old API you mean the v1, it is not really compatible with Python, so be careful when comparing

RyanDavies19, 2023-03-09:
Okay, I can try to workout CMake for that. By old API I meant using MoorDyn.cpp as opposed to MoorDyn2.cpp within v2. So far everything has been identical between the two which isn't surprising but I am including it to keep things thorough, and because I can use the same script for the dev2 branch as well.

commit 738f8682 (2023-03-24): Merge pull request #55 from sanguinariojoe/setup_py

Fix #54
closed 2023-03-24 by sanguinariojoe