FloatingArrayDesign/MoorDyn #62: Familiarization with Moordyn V2

timmaflan, 2023-04-04:
Hello! I am trying to implement a simple mooring system in moordyn v2. 

I am trying to attach a mooring line to a point absorber and attach it to a weight on the sea floor. From there, I am trying to attach the weight to two anchor points on the sea floor. This exercise is a simple test in understanding how rods, bodies, and connections work before moving on to a complex, and more realistic geometry. 
[image]

The first question I have is: can you use bodies without using rods? from the document https://www.nrel.gov/docs/fy20osti/76555.pdf, figure 2, page 3 I assume that I need rods to implement bodies, however it isn't immediately clear to me how they connect. 

Second: At the moment, I am having this error appear. Moordyn says it does not recognize "free" which I am using because of the documentation found [here](https://github.com/FloatingArrayDesign/MoorDyn/blob/v2/tests/Mooring/BodiesAndRods.dat). Any suggestions here?

[image]

My moordyn input file is also listed below. 

Mooring line data file for MoorDyn in Lines.dll
---------------------- LINE DICTIONARY -----------------------------------------------------
LineType   Diam    MassDenInAir    EA      BA/-zeta   Can Cat Cdn Cdt
(-)        (m)       (kg/m)        (N)      (Pa-s/-)  (-) (-) (-) (-)
LCP    0.020      0.319   5.617E+06   -1.0  1.0  0.00 1.00  0.0
---------------------- ROD TYPES ------------------------------------
TypeName      Diam     Mass/m    Cd     Ca      CdEnd    CaEnd
(name)        (m)      (kg/m)    (-)    (-)     (-)      (-)
Can          10       1.0e3     0.6    1.0     1.2      1.0
---------------------------- BODIES -----------------------------------------------------
ID   Attachment  X0     Y0    Z0     r0      p0     y0     Mass  CG*   I*      Volume   CdA*   Ca
(#)     (-)      (m)    (m)   (m)   (deg)   (deg)  (deg)   (kg)  (m)  (kg-m^2)  (m^3)   (m^2)  (-)
1       Free      0     0      0     0       0      0      1e6     0    1e10        0       0      0
---------------------- RODS ----------------------------------------
ID   RodType  Attachment  Xa    Ya    Za    Xb    Yb    Zb   NumSegs  RodOutputs
(#)  (name)    (#/key)    (m)   (m)   (m)   (m)   (m)   (m)  (-)       (-)
1      Can      Body1      0     0     2     0     0     15   3         p
2      Can   Body1Pinned   2     0     2     5     0     15   3         p
---------------------- NODE PROPERTIES -----------------------------------------------------
Node  Type    X      Y      Z      M     V    FX   FY   FZ   CdA  CA
(-)   (-)    (m)    (m)    (m)    (kg) (m^3) (kN) (kN) (kN) (m^2) (-)
1    Vessel  0.000 0.000 -18.00   0     0     0    0    0    0    0
2    Free    0.000 0.00  -65.00   0     0     0    0    0    0    0
3    Fix     0.000 -25.0 -65.00   0     0     0    0    0    0    0
4    Fix     0.000 -25.0 -65.00   0     0     0    0    0    0    0
---------------------- LINE PROPERTIES -----------------------------------------------------
Line LineType UnstrLen NumSegs NodeAnch NodeFair Flags/Outputs
(-)    (-)      (m)      (-)     (-)      (-)       (-)
1      LCP      47        3      2        1         tp
2      LCP      25        3      3        2         tp
3      LCP      25        3      4        2         tp
---------------------- SOLVER OPTIONS-----------------------------------------
0.0005   dtM - time step to use in mooring integration
0        WaveKin - wave kinematics flag (0=neglect, the only option currently supported)
6.0e7    kBot - bottom stiffness
6.0e7    cBot - bottom damping
65.0     WtrDpth - water depth
5.0      CdScaleIC - factor by which to scale drag coefficients during dynamic relaxation IC gen
0.001    threshIC - threshold for IC con
0        WriteUnits - option to skip units line in output files if zero
-------------------------- OUTPUTS --------------------------------
FairTen1
AnchTen1
FairTen2
AnchTen2
--------------------- need this line ------------------

sanguinariojoe, 2023-04-05:
My two cents:

1.- You can create coupled/free bodies and attach connections to them. See https://github.com/FloatingArrayDesign/MoorDyn/blob/v2/tests/Mooring/BodiesAndRods.dat#L24 . No need of rods

2.- Regarding the error. In general you should paste the output text rather than a screenshot. Anyway, if you read the output from the beginning you can see you are running MoorDyn v1, not v2

timmaflan, 2023-04-06:
1. As I stated, I was using the BodiesAndRods.dat file to influence my moordyn file. However, in the  BodiesAndRods.dat file, I don't understand what the "rods" and "rod Types" section adds to the simulation. Is it supposed to be attached to the mooring somewhere? Also, what does (Xa Ya Za Xb Yb Zb) mean? what do the subscripts a and b mean in this context? 
3. Thank you for your input. I will update to v2. 
Thanks.

timmaflan, 2023-04-06:
That's very helpful to know. I have a very robust wecSim model set up for the WEC. Are you also suggesting that the weight, located at (0,0,-65) needs to have a motion solver? 
Thanks for your help!

sanguinariojoe, 2023-04-06:
Nope, that one as free would work just fine. But you probably want to set some weight, e.g.

2 Free 0.000 0.00 -65.00 **15** 0 0 0 0 0 0

BTW, 3 segments per line is way too low. Increase that. Along this line, the numerical stability (and thus the maximum time step) is directly associated to the segments lenght. Thus, try to get the same segment length for each line. For instance 1 segment per meter:

 1 LCP 47 **47** 2 1 tp
2 LCP 25 **25** 3 2 tp
3 LCP 25 **25** 4 2 tp

mattEhall, 2023-04-27:
Hi @timmaflan, sorry, I'm only just looking at this now. Are things working okay now? 
When the line types are the same, keeping the same segment length is often a nice idea, but personally I'd only go down to 1 m segments if you need a really high level of accuracy. There's no harm in using fewer if you want to speed things up.
It's fine to have no mass on that free connection point at the bottom, if that's the scenario. 
Matt

sanguinariojoe, 2023-06-20:
I am closing this