FloatingArrayDesign/MoorDyn #70: Error introduced in bodies_and_rods test

djanders19, 2023-05-10:
Our team has observed that d0b09d27 causes `bodies_and_rods` to no longer converge during the IC stage. Furthermore, when MOORDYN_SSE is set to AVX2, the system actually diverges completely at `t=0.0003`s. This divergence is first caught in Line1, which connects Body1 and a fixed point. The two rods in the model appear to be linked to Body1 as well. This bug also persists in the latest commit ebcac61d.

We haven't determined the root cause of this, but it appears to be due to the changes to the mathematics in Misc.cpp, Misc.hpp, or Rod.cpp, as the input file was not changed in this commit. The issue persists even when the time-step is reduced to be quite low (0.000001), so it does not appear to be a minor stability thing.

AlexWKinley, 2023-05-11:
To expand on this, it seems like the culprit for the early divergence (introduction of NaNs) was this code starting at line 912 of Rod.cpp
```c++
vec r_top, r_bottom;
real zeta_i;
if (r[N][2] > r[0][2]) {
	r_top = r[N];
	r_bottom = r[0];
	real zeta_i = zeta[N];
} else {
	r_top = r[0];
	r_bottom = r[N];
	real zeta_i = zeta[0];
}

if ((r_bottom[2] < zeta_i) && (r_top[2] > zeta_i)) {
	// the water plane is crossing the rod
	// (should also add some limits to avoid near-horizontals at some point)
	h0 = (zeta_i - r_bottom[2]) / fabs(q[2]);
} else if (r[0][2] < zeta_i) {
	// fully submerged case
	h0 = UnstrLen;
} else {
	// fully unsubmerged case (ever applicable?)
	h0 = 0.0;
}
```
The variable `zeta_i` is not actually ever being initialized, causing it to be read from random uninitialized memory.  Fixing this problem does appear to solve the issue of the simulation quickly running into NaNs, but does not solve the problem of failing to converge during initial conditions solving.

sanguinariojoe, 2023-05-12:
Uhm, interesting! The tests were working just fine on my computer and [GitHub actions](https://github.com/FloatingArrayDesign/MoorDyn/actions/runs/4765512137/jobs/8471332141)

I will give it a look ASAP

AlexWKinley, 2023-05-12:
With standard settings the tests will pass, but what won't happen is reaching a steady state during the dynamic relaxation/initial conditions solving process. In the github actions log it looks like bodies_and_rods ran for 65 seconds, which at least in local testing generally means that it did not reach it's 1.5% tolerance during the dynamic relaxation and hit the ICmax. In previous commits the test normally finds a steady state rather quickly and would run for a much shorter amount of time.

RyanDavies19, 2023-05-24:
I'm not sure if I can say which approach is better, but in dev2 the rotation order was YXZ to deal with the FAST coordinate system convention (line 904 of Misc.cpp). In v2 the rotation order is XYZ (line 1046 of Misc.hpp). It might be worth trying to change the rotation matrix in v2 to YXZ and see if that resolves the issue.

sanguinariojoe, 2023-05-25:
Hey guys!

Today I will be visiting this issue and letting you know. Anyway, as a matter of fact, bodies_and_rods had some stability issues from the very beginning. In fact, @mattEhall rolled it out to me because it was not working in a previous version.

I hope I can come with news later today

mattEhall, 2023-05-25:
Hm.  The commit has changes in both the rotations and the moments of inertia.

In Rod::doRHS there are various changes to the moment of inertia calculations - both the magnitudes and which axis is which. I really think some explanation should be given in the commit message when the physics are being changed. Ryan, if you’re willing, maybe you could cross reference with the Fortran code. I don’t remember the calculations offhand, so I’m not sure if the commit in question is inserting deviations from the original physics, or if things had already deviated in earlier commits.

sanguinariojoe, 2023-05-25:
Ok, I have added a macro to define all the Euler rotations:

https://github.com/sanguinariojoe/MoorDyn/blob/fix70/source/Misc.hpp#L729

Then I can replace the calls to RotXYZ by RotZYX (that was the way it was working before the commit). i.e. on Body.cpp:

```
diff --git a/source/Body.cpp b/source/Body.cpp
index f665d43..2148da7 100644
--- a/source/Body.cpp
+++ b/source/Body.cpp
@@ -107,7 +107,7 @@ Body::setup(int number_in,
        v6 = vec6::Zero();
 
        // calculate orientation matrix based on latest angles
-       OrMat = RotXYZ(r6(Eigen::seqN(3, 3)));
+       OrMat = RotZYX(r6(Eigen::seqN(3, 3)).reverse());
 
        LOGDBG << "Set up Body " << number << ", type " << TypeName(type) << ". "
               << endl;
@@ -360,7 +360,7 @@ Body::updateFairlead(real time)
                v6 = rd_ves;
 
                // calculate orientation matrix based on latest angles
-               OrMat = RotXYZ(r6(Eigen::seqN(3, 3)));
+               OrMat = RotZYX(r6(Eigen::seqN(3, 3)).reverse());
 
                // set positions of any dependent connections and rods
                setDependentStates();
@@ -379,7 +379,7 @@ Body::setState(vec6 pos, vec6 vel)
        v6 = vel;
 
        // calculate orientation matrix based on latest angles
-       OrMat = RotXYZ(r6[3], r6[4], r6[5]);
+       OrMat = RotZYX(r6[5], r6[4], r6[3]);
 
        // set positions of any dependent connections and rods
        setDependentStates();
```

And on Rod.cpp:

```
diff --git a/source/Rod.cpp b/source/Rod.cpp
index 5208f14..3048cd4 100644
--- a/source/Rod.cpp
+++ b/source/Rod.cpp
@@ -364,12 +364,12 @@ Rod::setState(vec6 pos, vec6 vel)
        if (type == FREE) {
                // end A coordinates & Rod direction unit vector
                r6(Eigen::seqN(0, 3)) = pos(Eigen::seqN(0, 3));
-               const mat OrMat = RotXYZ(pos(Eigen::seqN(3, 3)));
+               const mat OrMat = RotZYX(pos(Eigen::seqN(3, 3)).reverse());
                r6(Eigen::seqN(3, 3)) = OrMat * q0;
                // end A velocities & rotational velocities
                v6 = vel;
        } else if ((type == CPLDPIN) || (type == PINNED)) {
-               const mat OrMat = RotXYZ(pos(Eigen::seqN(3, 3)));
+               const mat OrMat = RotZYX(pos(Eigen::seqN(3, 3)).reverse());
                r6(Eigen::seqN(3, 3)) = OrMat * q0;
                v6(Eigen::seqN(3, 3)) = vel(Eigen::seqN(3, 3));
        } else {
```

So it seems this change:

[image]

is not to be blamed

mattEhall, 2023-05-25:
Okay. That sounds like a nice flexible thing to have.
Do you remember why you changed the calculations of I_r and I_l, and changed what directions they were assigned to?

sanguinariojoe, 2023-05-25:
Yes, I do. Now the rods various computations are done in its local coordinates, i.e. the rod is oriented along the positive x axis. Afterwards the rotations are applied.

We can of course change that criteria to assume local coordinates on z direction. I just noticed the implementation seemed to work better considering x instead (but later on I changed a bunch of other things, so who knows if I was really right or not)

sanguinariojoe, 2023-05-25:
If I revert this change:

[image]

(looking at the paper it seems to me the minus is ok though) then the test is never reaching the asked accuracy, but it is not clearly converging:

Remaining error after 10.01 s = 3.11306%
Remaining error after 20 s = 2.65188%
Remaining error after 30 s = 2.92492%

And rods test is still working just fine. Thus I am reverting that change.

Now I am playing a bit with the bodies_and_rods parameters

sanguinariojoe, 2023-05-25:
You gonna love this guys!

[image]

It might be a bug, but it is so beautiful! :-P

sanguinariojoe, 2023-05-25:
OK, so these are the facts I have right now:

 - The implementation of rods before d0b09d27993287c5d6788242a3c89efbf9ac450c was not working. Even simplest tests like https://github.com/FloatingArrayDesign/MoorDyn/blob/v2/tests/rods.cpp were failing. The rods were rotating around random axes. This was caused by 2 errors:
   - Wrong inertia and momenta computation
   - Ill time integration because we were summing up angular velocities to direction vectors
 - The rods after 949e9afbcc636b88b7314c04ba823a503dd7ebc0 seems to work fine, and interacts correctly with lines
 - The test bodies_and_rods broke after d0b09d27993287c5d6788242a3c89efbf9ac450c
 - If I remove the rods from the test (bodies_and_rods), the test is working (the body is just sinking until it reaches the seabed)
 - If I change the test putting the rods facing downwards:
```
1      Can      Body1      0     0     -2     0     0     -15   3         p
2      Can   Body1Pinned   2     0     -2     5     0     -15   3         p
```
Then the test is working

It seems the problem happens at the time instant the rods touch the free surface (from above water). I continue checking

AlexWKinley, 2023-05-25:
I don't have clear answers for what the problem is but I do have a few notes about rods especially near the surface that may be useful.

Rod 2 in the bodies and rods test is not actually pinned to body1 like it claims to be, because "Body1Pinned" is more than 9 characters it gets cut off and ends up being fixed to body 1 (i have a pull request in the works that will fix this problem in general but changing it to `Body1Pin` will produce the correct behavior in the mean time.

The water plane moment of inertia calculation is entirely incorrect (looking at the fortran code it looks like it was changed there to just be stable). But the biggest reason for that is that `phi` as defined in the rod is angle relative to horizontal, while the equation used for water plane moment assumes phi to be the angle relative to vertical. I also think that more generally the water plane moment of rods needs more careful consideration or at least safeguards if things like rods floating sideways in the water are to be handled properly.

I think there are some other issues with how rods work especially near the surface, I haven't fully figured out all of the problems but I've had tests end up with all sorts of nonphysical behavior that happens seemingly out of nowhere when some unknown degenerate case is hit that leads to incorrectly calculated very large forces. I seem to recall that part of the issue may be buoyancy calculations, if memory serves I've had tests calculate nonsense buoyancy values for partially submerged rods.

sanguinariojoe, 2023-05-26:
Ok, for the time being I merged [this](https://github.com/FloatingArrayDesign/MoorDyn/pull/73) on v2, which is reverting the order of rotations for bodies, and fixing some minor bugs introduced at https://github.com/FloatingArrayDesign/MoorDyn/pull/71

I also corrected getH as you suggested @AlexWKinley

sanguinariojoe, 2023-05-26:
P.S. It seems @AlexWKinley  is on the track. I am quite busy with another project though, so I let him do his magic.

Do not doubt on summoning me if I can be of any use!

mattEhall, 2023-05-26:
Thanks for looking into things guys, and for finding those additional issues Alex. As you point out, the Rod hydrostatic calculations are set up for measuring rotations relative to vertical. So it’s a hassle if we decide to suddenly compute things relative to a horizontal axis instead. 

If there are concerns with the surface-piercing computations, we should update to match the FORTRAN version—its surface-piercing behavior works very well.

I’m a little surprised how many additional errors have been introduced (for example the missing minus signs in getH). I don’t remember what the exact status was before the refactoring began, but I’m pretty sure I had most basic things with Rods working (I used them in some papers and projects). To clear this up maybe it’s best if we go through and revise/revert everything to be consistent with what’s currently in FORTRAN (most of which would match the pre-refactored version).

Thoughts?

sanguinariojoe, 2023-05-26:
+1 !!!

Let's mimic the Fortran version and check again. I suppose it will work. Otherwise we are fixing 2 codes in one shot

RyanDavies19, 2023-06-26:
Following up here to say that I just submitted a PR that is the first round of matching the FORTRAN rod dynamics. There should be more coming soon, and further down the road bodies as well.

PR #90 (2023-07-07): Bodies and Rods Fixes
# Fixing Bodies and Rods

These changes should fix most (hopefully all) of the major issues with bodies and rods working properly. Closes #70 

The biggest change to MoorDyn is an overhaul of the kinematics for the 6DOF objects, bodies and rods. 
The full math of this change is detailed in the 6DOF kinematic section below. 
The overall highlights are that instead of euler angles, bodies and rods now use quaternions to represent their rotation. 
The integration of angular velocity in MoorDyn turned out to be incorrect (see 6DOF kinematics section), and this has been addres [... cut]
closed 2023-09-12 by sanguinariojoe