FloatingArrayDesign/MoorDyn #72: Clang++ compile issue with Waves

RyanDavies19, 2023-05-25:
I was getting my branch up to date today and merged all the recent updates in. I have been compiling the library from the makefile in compile/DYLIB/makefile. v2 no longer compiles, and clang throws an error: linker command failed with exit code 1. It says this is caused by undefined symbols for architecture x86_64, all of which are related to the waves object. @sanguinariojoe, any ideas?

[image]

sanguinariojoe, 2023-05-26:
Damn! Makefiles! I hate you!

Now it should compile properly. Let me know please

RyanDavies19, 2023-05-26:
That fixed it, thank you!

closed 2023-05-26 by RyanDavies19