FloatingArrayDesign/MoorDyn #76: How to control a 'Coupled' point by calling program？

ldd289806329, 2023-06-06:
I want to control a certain point on the line and  it can move according to my calling program/script. But I only know the 'Coupled' point, which can be controlled by calling program, but I don't know how to implement it. There is no clear explanation in the documentation. Do I need to make modifications in the source program? Which source program can be modified to achieve control over  the 'Coupled' point ?thank you
[image]

sanguinariojoe, 2023-06-07:
I assume you are considering the version 2.

You can take a look to the tests. e.g. https://github.com/FloatingArrayDesign/MoorDyn/blob/v2/tests/quasi_static_chain.cpp#L201

As you can see, no need to change a single line of the source code. When you call [MoorDyn_Step()](https://moordyn-v2.readthedocs.io/en/latest/doxygen/group__new__c__api.html#ga5903afba6241f15dc1f2f660be5d2718) you are meant to pass the positions and velocities of all the coupled entities.

In your case you just need to pass a 3 components array x with the position of the point and another 3 components array xd with the velocity of the point.

closed 2023-06-07 by sanguinariojoe

ldd289806329, 2023-06-20:
Hello, I would like to use Moordyn for my project and may need to make some changes to its source code. It is a real-time hybrid test of the mooring system, which requires Moordyn to be made into a numerical substructure and an experimental substructure. The mooring line needs to be cut off from the middle, with a portion of the upper mooring line retained, and specific displacement or force applied at the cutoff point to simulate the true dynamic response of the mooring line in an unbroken mooring. The problem I am facing now is that I don't know how to apply displacement or force in real-time at the cutoff point, or how to control the endpoint of the mooring line in real-time through external programs such as MATLAB. I hope you can give me some suggestions or tell me where to modify the source code? The following is a schematic diagram of my project.thanks
[image]