FloatingArrayDesign/MoorDyn #82: Unit Testing Framework

AlexWKinley, 2023-06-20:
Kelson Marine is looking to add unit testing to MoorDyn. We want to find a framework/system to integrate with MoorDyn so that our internal tests can be shared publicly in this repo. There are a number of different unit testing frameworks and libraries for C++, and we would appreciate peoples thoughts and preferences.  @mattEhall @sanguinariojoe 

The two main questions are which library or framework we should use, and how we should integrate it into MoorDyn. 

For libraries there are a lot of options including
* GoogleTest
* Catch2
* Many other small and simple unit testing libraries

Our first instinct is probably GoogleTest, if for no other reason than the fact that the benchmarks already use the related Google benchmark library. But Catch2 or any of the header-only unit testing libraries available for C++ would also serve our needs perfectly well, so if people have preferences or opinions we are very open.

In terms of integrating the testing library into MoorDyn we would probably make a `external/` folder in the root of the repository and put the testing library inside of there and set up cmake to build it. Then continue to use the existing `tests/` folder and make additional .cpp files for the unit tests and set up the cmake to build those with the testing library and add them to the ctests. If there is a different preferred way to do this, we're happy alter the setup.

Let us know if you have any thoughts or ideas for ways this plan could be improved.

sanguinariojoe, 2023-06-21:
I really prefer something simple like Catch2. I indeed though about using a simple Catch2-like framework at the beginning. But since I was also setting up the CMake and so on, I just went with raw tests.

So my vote is for Catch2!

AlexWKinley, 2023-06-21:
I will say that for Catch2, it seems like since their 3.0 version they have moved away from being a header only library. I still agree that it seems like a good and practical choice and will likely be what we go with.

closed 2023-09-12 by sanguinariojoe