FloatingArrayDesign/MoorDyn #85: New compile issue with V2 makefile

RyanDavies19, 2023-06-22:
With the changes to the V2 makefile for Mac (in compile/DYLIB/), I am getting compile issues with Waves.h:

../../source/Waves/../Waves.h:68:13: error: expected ';' after top level declarator
        int DECLDIR MoorDyn_GetWavesKin(MoorDynWaves waves,
                   ^
                   ;
../../source/Waves/../Waves.h:85:9: error: redefinition of 'DECLDIR' with a different type: 'double' vs 'int'
        double DECLDIR WaveNumber(double Omega, double g, double h);

I am not super familiar with how the new makefile works, and can't figure out why the compiler is not recognizing DECLDIR.

sanguinariojoe, 2023-06-26:
I would say this has very little to do with the makefile. Does it work with CMake?

BTW, paste the full log, so we can see which source file (.cpp) is creating the issue

sanguinariojoe, 2023-06-26:
This is probably fixing the issue:

https://github.com/FloatingArrayDesign/MoorDyn/pull/86

Try it out and let me know please

RyanDavies19, 2023-06-26:
That resolved it, it is compiling fine now. Thanks for the help

sanguinariojoe, 2023-06-26:
OK, merged then!

closed 2023-06-26 by sanguinariojoe