FloatingArrayDesign/MoorDyn #104: Updating docs

RyanDavies19, 2023-08-14:
We are working on updating the MoorDyn docs. @sanguinariojoe or @AlexWKinley, do you know if the cmake compile method still needs Eigen installed separately? I know it comes with the repo now so I was wondering if the instructions below were outdated.

https://github.com/FloatingArrayDesign/MoorDyn/blob/6395ced8fefcd8d3457bd15637b4359914457cfe/docs/starting.rst?plain=1#L137

sanguinariojoe, 2023-08-14:
No, it is not outdated. Effectively MoorDyn can be compiled and installed
with the Eigen provided within the repo.

However, the Eigen copy distributed with MoorDyn cannot be safely
installed, because it will collide with the one distributed by each OS.

That renders the C++ library rather useless (missing headers errors will be
received).

Note: C, Python, Fortran and Rust are not affected.

RyanDavies19, 2023-08-14:
Thanks for the quick response, that sounds good. We will leave that in there then.

closed 2023-08-14 by RyanDavies19