FloatingArrayDesign/MoorDyn #105: Python Package install issue via CMake

RyanDavies19, 2023-08-15:
@sanguinariojoe I was trying to install the python package via the methods described in the docs, however I am getting an error from pip when running the setup.py file saying that the readme.md file can't be found (see attached screenshot). Any thoughts on how to resolve this? 

[image]

Second question, when using the cmake installation three different MoorDyn libraries are created in `MoorDyn/build/source/` : 
`libmoordyn.2.2.dylib	libmoordyn.2.dylib 	libmoordyn.dylib`

What is the difference between these three and which one should people be directed to use?

RyanDavies19, 2023-08-15:
@sanguinariojoe Is there a specific file name I should be looking for for those libraries that are in the different directory? 

Secondly, from a user perspective, which one of the dylib files should they be pointed towards using?

sanguinariojoe, 2023-08-16:
1.- libmoordyn.dylib

2.- All of them.

When you are compiling the -lmoordyn option tells the linker to look for
the libmoordyn.dylib file. That is called the development file.

After the binary is compiled, it is attached to the libmoordyn.2.2.dylib,
i.e. to make the binary work, the SO shall be able to find the file
libmoordyn.2.2.dylib

Now, let say we launch a new version, MoorDyn-2.3. If the user installs the
new version, the libraries will change from:

libmoordyn.dylib -> libmoordyn.2.dylib -> libmoordyn.2.2.dylib

To:

libmoordyn.2.2.dylib
libmoordyn.dylib -> libmoordyn.2.dylib -> libmoordyn.2.3.dylib

The binary will still work no matter what we changed, because
libmoordyn.2.2.dylib is still there.

The user is of course entitled to recompile, linking with the new
libmoordyn.dylib. The new binary is attached now to libmoordyn.2.3.dylib

RyanDavies19, 2023-08-16:
@sanguinariojoe thanks for that explanation. I managed to get the python package installed correctly, however it won't run in an external script because it can't locate the library. I installed locally because I don't have admin rights, below is the error that python gives when trying to run a script that has `import moordyn`.

`File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/moordyn/moordyn.py", line 75, in Create
    import cmoordyn
ImportError: dlopen(/Users/rdavies/Library/Python/3.9/lib/python/site-packages/cmoordyn.cpython-39-darwin.so, 0x0002): Library not loaded: @rpath/libmoordyn.2.dylib
  Referenced from: <7B8872E0-34F6-358B-97D8-0AC45028F713> /Users/rdavies/Library/Python/3.9/lib/python/site-packages/cmoordyn.cpython-39-darwin.so
  Reason: tried: '/usr/local/lib/libmoordyn.2.dylib' (no such file), '/usr/lib/libmoordyn.2.dylib' (no such file, not in dyld cache)
`

RyanDavies19, 2023-08-16:
I tried that and adding it to $HOME/.profile but neither worked. Which file is @rpath defined in? 

It seems like it would best if that was just set to the path where the libraries are compiled, i.e. MoorDyn/build/source/. Then from a user perspective they could easily know why something wasn't working.

RyanDavies19, 2023-08-16:
Moving the library didn't work either. I will play around with this and see what I can come up with. Thanks for the input

closed 2023-08-16 by RyanDavies19