FloatingArrayDesign/MoorDyn #106: error:cxx_variable_templates

dustelf, 2023-08-16:
Please,tell me why.
erro when using cmake.
**Using Precompiled Headers!
CMake Error at extern/Catch2/src/CMakeLists.txt:351 (target_compile_features):
  target_compile_features The compiler feature "cxx_variable_templates" is
  not known to CXX compiler

  "GNU"

  version 4.9.2.**

awkinley, 2023-08-16:
Catch2 is looking for some c++14 features to ensure that you are compiling with a c++14 compatible compiler. MoorDyn itself also makes use of some c++17 features I believe. More details on gcc's c++ feature support can be found here: https://gcc.gnu.org/projects/cxx-status.html.

In short, you just need to install a more recent c++ compiler, and you should be good to go. You may need to tell cmake to use the newer compiler. Instructions for that can be found at https://stackoverflow.com/questions/45933732/how-to-specify-a-compiler-in-cmake

dustelf, 2023-08-16:
I know..
I used the lower version od the default when cmake specified the gcc path,thank you

closed 2023-08-16 by dustelf