FloatingArrayDesign/MoorDyn #107: error in 'MoorDyn.build>mingw32-make'

dustelf, 2023-08-16:
/(ㄒoㄒ)/~~
Please tell me why.

E:/msys/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\bodies.dir/objects.a(bodies.cpp.obj):bodies.cpp:(.text+0x168e): undefined reference to `moordyn::Euler2Quat(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
collect2.exe: error: ld returned 1 exit status

sanguinariojoe, 2023-08-16:
Probably our fault. Please, provide us with the full list of commands you
typed and the logs/outputs

sanguinariojoe, 2023-08-17:
> 可能是我们的错。请向我们提供您命令的完整列表 类型和日志/输出
> 周三， 16 八月 2023， 08：10 dustelf， *@*.*> 写道： /（ㄒoㄒ）/~~ 请告诉我为什么。
> 未定义对 'moordyn：：Euler2Quat（Eigen：：Matrix<double， 3， 1， 0， 3， 1> 常量&）'
> collect2.exe：错误：LD 返回 1 个退出状态 — 直接回复此电子邮件，在 GitHub 上查看 <#107>
> . 您收到此消息是因为您订阅了此线程。消息 编号： @.*>
> I just followed your steps and entered this command on cmd
> "mingw32-make"/(ㄒoㄒ)/~~
> [ 88%] Built target bodies_and_rods
> [ 89%] Building CXX object tests/CMakeFiles/bodies.dir/bodies.cpp.obj
> [ 89%] Linking CXX executable bodies.exe
> mingw32-make[2]: *** [tests\CMakeFiles\bodies.dir\build.make💯
> tests/bodies.exe] Error 1
> mingw32-make[1]: *** [CMakeFiles\Makefile2:1322:
> tests/CMakeFiles/bodies.dir/all] Error 2
> mingw32-make: *** [Makefile:165: all] Error 2

I am out of the office the whole august. I will check this out ASAP.

dustelf, 2023-08-18:
> I am out of the office the whole august. I will check this out ASAP.
> […](#)

ok,ok

sanguinariojoe, 2023-09-04:
This is fixed after https://github.com/FloatingArrayDesign/MoorDyn/pull/111

closed 2023-09-04 by sanguinariojoe