FloatingArrayDesign/MoorDyn #108: Total Force and Moment Calculation Issue

soydanahmet, 2023-08-16:
Hi,

I'm trying to couple Mordyn v1.01.02 with my code and my input file is attached. My platform center of mass is 3.5, 0.5, 0.4744.

So X=3.5,0,5,0.4744 and XD=0.0,0.0,0.0 at the beginning.

Basically, I initialize Moordyn by using LinesInit(X, XD) and then I call LinesCalc(X, XD, Flines, &tprev, &deltaT) function for force calculation.

In the first iteration, I suppose to get zero force along the x and y directions (Fx, Fy) from MoorDyn but I get some forces. (In the first iteration X and XD are the same as at the beginning). So what would be the reason that in the first iteration, I get the wrong force result from MoorDyn?

Thanks in advance.

[lines.txt](https://github.com/FloatingArrayDesign/MoorDyn/files/12358280/lines.txt)

sanguinariojoe, 2023-08-16:
My two cents, do not use MoorDyn v1

RyanDavies19, 2023-08-16:
Hi, MoorDyn v1 is the older version of the code, v2 is the most up to date version. v1 should still work for your case though, as it is only lines and connections. Even though your points aren't moving MoorDyn will calculate the forces from the line sinking and from the weight of the lines. Likely the forces you see on the fairlead and anchor points are coming from these line tensions. Does that make sense for the order of magnitude of what you are seeing?

sanguinariojoe, 2023-08-17:
> Here are the forces and moments I get from MoorDyn v1 by using
> LinesCalc(X, XD, Flines, &tprev, &deltaT) function.
>
> Since I use 4 symmetrical mooring lines, I'd expect that Fx and Fy forces
> would be zero and only get Fz. However, Fx and Fy forces don't look
> plausible.
>
> Fx: -170.27
> Fy: -22.397
> Fz: -41.366
> Mx: -11.069
> My: 71.12
> Mz: 7.1348

Many errors during the initialization has been fixed since version 1, not
to mention the memory errors which make the things quite random sometimes.

Would you try the version 2 and let us know?

soydanahmet, 2023-08-18:
Now I'm able to get reasonable forces with version 2 only if I moved the center of the body to 0,0,0 and create the input file accordingly. If I don't move it, the forces remain as before.

But now, I have another confusion and would very much appreciate it if you can clarify this:

In the method below, X stores the center of mass and the rotation angles of the body. As far as I understand, the unit of rotation angles should be rad/s and the first angle X[3] should be the roll angle, the second  X[4 ] pitch, and the third one X[5] yaw angle. Am I right?

MoorDyn_Step(moordyn_, X, XD, Flines, &tprev, &deltaT);

Thank you.

RyanDavies19, 2023-08-18:
You are correct with the order being roll, pitch, yaw, however the units are degrees. Also just a side note, in the v1 input file you shared you had all of your points above the waterline. Not sure if that was your intention. In MoorDyn the waterline is assigned z = 0 and objects beneath it have negative z values.

AlexWKinley, 2023-08-18:
I believe that body rotation is expected to be in radians. I remember looking at it during the quaternion work and it never did the degrees to radians conversion. And a quick skim of the current code doesn't seem to do that conversion either. At the very least I'm fairly certain that since that work has happened, it expects radians.

RyanDavies19, 2023-08-18:
@AlexWKinley you're right. I had thought it had been changed but that was only for outputs. I chatted with @mattEhall about it and we think inputs in degrees are best. I'll do a PR today that adjusts that for rods and bodies so that inputs are degrees as well.

sanguinariojoe, 2023-08-19:
> Now I'm able to get reasonable forces with version 2 only if I moved the
center of the body to 0,0,0 and create the input file accordingly. If I
don't move it, the forces remain as before.

Keep in mind that the coordinates of any point attached to the body shall
be set as relative to such body. I.e. subtract the body center (I assume
that the body has not initial angles) to the points 5-8

closed 2023-09-08 by sanguinariojoe