FloatingArrayDesign/MoorDyn #118: using MoorDyn with matalb----time step calculate

Echo-new, 2023-09-12:
Dear sir
I have two questions about using MoorDyn with matalb wrapper. 
First, in your typical example, it shows how to calculate time step when the fairlead moves at a constant speed, but I wonder if the fairlead does sinusoidal motion, how to change the code?
[image]
When I force 5sin(10t) , it will be changed :
[image]
But the result was not satisfying, my first x=[0 0 -70]’, calculate one step x=[0 0 -70]’+5*sin(10*dt)*[0 0 1]'+dx*dt. It seems that the program run the step dx*dt by default. So I wondered what should I do?
Second, when I set time step(dt) from 0.001s to 1s in matlab, it gives near totally same answer . So I want to know is it because Runge Kutta changes the time step?

sanguinariojoe, 2023-09-12:
First, please do not paste code screenshots, it makes harder to discuss it. Just copy the code and use the "<>" tool on the text editor.

`MoorDynM_Step` linearly integrates the position of the fairlead from `x` (at the beginning of the time step, t) to `x + dx*dt` (at the end of the time step, t+dt). That is indeed the expected behavior. Probably your problem is that you are computing dx(t) instead of dx(t + 0.5*dt), which is naturally more consistent. Or you can even compute dx = (x(t+dt) - x(t)) / dt, which will give you the best possible approach.

> Second, when I set time step(dt) from 0.001s to 1s in matlab, it gives near totally same answer . So I want to know is it because Runge Kutta changes the time step?

No

Echo-new, 2023-09-12:
Thank you for your reply.
I understand the first answer. But for the second question, I mean that perhapes we use Runge Kutta as an adaptive time step method, because that time step (dt) cannot really be 1second. And I see the following in source code MoorDyn2.cpp:
<//default value for desired mooring model time step
dtM0=0.001>
So I guess it has its own calculating time step in program. Right?

RyanDavies19, 2023-09-12:
@Echo-new chiming in to offer some time step clarity. The dt value you give to the step function needs to be the same value that is specified in the input file as dtM. This is what determines the internal tilmestep of the model. The default value you mention is the value that MoorDyn uses if no dtM is specified in the input file. You can read more in the inputs section of the documentation.

Echo-new, 2023-09-14:
Thank you, sir.
I think that I get it. It confused me that why I set dt from 0.01 to 1 in matlab, can give same position datas. I know now that dtM is the true intergration time step which can be changed in input file. Thank you again.

RyanDavies19, 2023-09-14:
Happy to help

closed 2023-09-14 by RyanDavies19