FloatingArrayDesign/MoorDyn #120: Error in running C++/ Linux

alda30, 2023-09-14:
Dear,

I am trying to use Moordyn package using C++ on my Ubuntu machine. 
When I try to run an example file I by typing the commad "g++ file.cpp"

I get this:
```
In file included from /usr/local/include/moordyn/IO.hpp:37,
                 from /usr/local/include/moordyn/MoorDyn2.hpp:38,
                 from try.cpp:2:
/usr/local/include/moordyn/Misc.hpp:171:16: error: ‘Vector’ in namespace ‘Eigen’ does not name a template type; did you mean ‘VectorXi’?
  171 |         Eigen::Vector<real, 7> toVec7() const
      |                ^~~~~~
      |                VectorXi
In file included from /usr/local/include/moordyn/Time.hpp:40,
                 from /usr/local/include/moordyn/MoorDyn2.hpp:41,
                 from try.cpp:2:
/usr/local/include/moordyn/Line.hpp: In member function ‘void moordyn::Line::getFASTtens(float*, float*, float*, float*) const’:
/usr/local/include/moordyn/Line.hpp:476:52: error: ‘seqN’ is not a member of ‘Eigen’
  476 |                 *FairHTen = (float)(Fnet[N](Eigen::seqN(0, 2)).norm());
      |                                                    ^~~~
/usr/local/include/moordyn/Line.hpp:478:52: error: ‘seqN’ is not a member of ‘Eigen’
  478 |                 *AnchHTen = (float)(Fnet[0](Eigen::seqN(0, 2)).norm());
      |                                                    ^~~~
In file included from /usr/local/include/moordyn/Time.hpp:43,
                 from /usr/local/include/moordyn/MoorDyn2.hpp:41,
                 from try.cpp:2:
/usr/local/include/moordyn/Body.hpp: In member function ‘moordyn::vec moordyn::Body::getVelocity() const’:
/usr/local/include/moordyn/Body.hpp:310:59: error: ‘seqN’ is not a member of ‘Eigen’
  310 |         inline vec getVelocity() const { return v6(Eigen::seqN(0, 3)); }
      |                                                           ^~~~
/usr/local/include/moordyn/Body.hpp: In member function ‘moordyn::vec moordyn::Body::getAngularVelocity() const’:
/usr/local/include/moordyn/Body.hpp:315:66: error: ‘seqN’ is not a member of ‘Eigen’
  315 |         inline vec getAngularVelocity() const { return v6(Eigen::seqN(3, 3)); }
      |                                                                  ^~~~
In file included from /usr/include/Eigen/Core:22,
                 from /usr/include/Eigen/Dense:1,
                 from /usr/include/Misc.hpp:37,
                 from /usr/include/Waves/SpectrumKin.hpp:3,
                 from /usr/local/include/moordyn/Waves.hpp:43,
                 from /usr/local/include/moordyn/MoorDyn2.hpp:42,
                 from try.cpp:2:
/usr/include/Eigen/src/Core/util/ConfigureVectorization.h: At global scope:
/usr/include/Eigen/src/Core/util/ConfigureVectorization.h:461:27: error: redefinition of ‘const char* Eigen::SimdInstructionSetsInUse()’
  461 | inline static const char *SimdInstructionSetsInUse(void) {
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/Eigen/Dense:1,
                 from /usr/local/include/moordyn/Misc.hpp:37,
                 from /usr/local/include/moordyn/IO.hpp:37,
                 from /usr/local/include/moordyn/MoorDyn2.hpp:38,
                 from try.cpp:2:

```

Could someone please help?

AlexWKinley, 2023-09-14:
It seems like the version of Eigen installed on your system is older than what MoorDyn needs. To solve this you could either install the latest version of Eigen on your system (I believe any 3.4.xx version should work), or what will also probably work is compiling MoorDyn with the cmake flag "-DEXTERNAL_EIGEN=OFF", which will use the copy of Eigen in MoorDyn. If you download the eigen source [http://eigen.tuxfamily.org/index.php?title=Main_Page](http://eigen.tuxfamily.org/index.php?title=Main_Page), it should have instructions for the straightforward install procedure (nothing needs to be compiled as it's a header only library).

sanguinariojoe, 2023-10-04:
He is not replying, so I assume everything is working fine

closed 2023-10-04 by sanguinariojoe