FloatingArrayDesign/MoorDyn #121: Local docs compiling

RyanDavies19, 2023-09-14:
@sanguinariojoe do you have any idea what could be causing this error in compiling the documentation locally:

`Patching output file 56/58
Patching output file 57/58
Patching output file 58/58
type lookup cache used 10518/65536 hits=122962 misses=10670
symbol lookup cache used 6206/65536 hits=83658 misses=6206
finished...
/Users/rdavies/Library/Python/3.9/lib/python/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
/Users/rdavies/Library/Python/3.9/lib/python/site-packages/breathe/project.py:116: RemovedInSphinx80Warning: Sphinx 8 will drop support for representing paths as strings. Use "pathlib.Path" or "os.fspath" instead.
  self._default_build_dir = os.path.dirname(app.doctreedir.rstrip(os.sep))
making output directory... done

Exception occurred:
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/re.py", line 210, in sub
    return _compile(pattern, flags).sub(repl, string, count)
TypeError: expected string or bytes-like object
The full traceback has been saved in /var/folders/46/4_pcs77x31j8w360rc7pfv6chrj_sl/T/sphinx-err-athjs7ze.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
make[2]: *** [docs/sphinx/index.html] Error 2
make[1]: *** [docs/CMakeFiles/Sphinx.dir/all] Error 2
make: *** [all] Error 2`

The traceback that was saved shows the following:
`# Platform:         darwin; (macOS-13.5.2-x86_64-i386-64bit)
# Sphinx version:   7.2.6
# Python version:   3.9.6 (CPython)
# Docutils version: 0.18.1
# Jinja2 version:   3.1.2
# Pygments version: 2.15.1

# Last messages:

# Loaded extensions:

# Traceback:
Traceback (most recent call last):
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/sphinx/cmd/build.py", line 293, in build_main
    app = Sphinx(args.sourcedir, args.confdir, args.outputdir,
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/sphinx/application.py", line 272, in __init__
    self._init_builder()
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/sphinx/application.py", line 342, in _init_builder
    self.builder.init()
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/sphinx/builders/html/__init__.py", line 219, in init
    self.init_templates()
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/sphinx/builders/html/__init__.py", line 270, in init_templates
    self.theme = theme_factory.create(themename)
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/sphinx/theming.py", line 227, in create
    self.load_extra_theme(name)
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/sphinx/theming.py", line 181, in load_extra_theme
    self.load_external_theme(name)
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/sphinx/theming.py", line 194, in load_external_theme
    theme_entry_points = entry_points(group='sphinx.html_themes')
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/importlib_metadata/__init__.py", line 933, in entry_points
    return EntryPoints(eps).select(**params)
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/importlib_metadata/__init__.py", line 930, in <genexpr>
    eps = itertools.chain.from_iterable(
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/importlib_metadata/_itertools.py", line 16, in unique_everseen
    k = key(element)
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/importlib_metadata/_py39compat.py", line 18, in normalized_name
    return dist._normalized_name
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/importlib_metadata/__init__.py", line 855, in _normalized_name
    or super()._normalized_name
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/importlib_metadata/__init__.py", line 473, in _normalized_name
    return Prepared.normalize(self.name)
  File "/Users/rdavies/Library/Python/3.9/lib/python/site-packages/importlib_metadata/__init__.py", line 773, in normalize
    return re.sub(r"[-_.]+", "-", name).lower().replace('-', '_')
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/re.py", line 210, in sub
    return _compile(pattern, flags).sub(repl, string, count)
TypeError: expected string or bytes-like object`

sanguinariojoe, 2023-09-15:
> 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: urllib3/urllib3#3020

Pretty eclectic error, but it seems to me that you do not have the theme
installed:

sphinx_rtd_theme

As far as I remember it shall be installed with pip

RyanDavies19, 2023-09-15:
@sanguinariojoe thanks, I was curious if you had ideas. Turns out when I installed Xcode with admin rights for something else, it created a second python copy. I had to go in and delete all that but things are working well now.

closed 2023-09-15 by RyanDavies19