FloatingArrayDesign/MoorDyn #122: Error installing Python version of MoorDyn (the stand-alone version)

tlopez-olocco, 2023-09-15:
I was trying to install the standalone version of moordyn to run some test without coupling the wind turbine.

This is what I do:

python -m pip install moordyn  or pip install moordyn 

Here is the error message I get: 

_**Collecting moordyn
  Using cached moordyn-2.2.0.tar.gz (112 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [9 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\tomil\AppData\Local\Temp\pip-install-xd0vc_s2\moordyn_48fef3810d494161bee4d25c744f82c3\setup.py", line 64, in <module>
          shutil.copytree(os.path.join('wrappers', 'python', 'moordyn'), 'moordyn')
        File "C:\Users\tomil\AppData\Local\Programs\Python\Python311\Lib\shutil.py", line 559, in copytree
          with os.scandir(src) as itr:
               ^^^^^^^^^^^^^^^
      FileNotFoundError: [WinError 3] The system cannot find the path specified: 'wrappers\\python\\moordyn'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.**_

I am wondering, is it maybe because I am missing some python packages installed? Or it may be something else? Any additional files from the github repository I might have to have in some kind of path?

Thanks in advance!

RyanDavies19, 2023-09-15:
@tlopez-olocco when you are running MoorDyn with a wind turbine are you using the version in OpenFAST? The best way to test that version is by using the MoorDynF driver that ships with OpenFAST. It allows you to run MoorDynF stand alone. Instructions can be found [here](https://moordyn.readthedocs.io/en/latest/inputs.html#moordyn-f-driver-input-file), and once compiled it is called by executing `moordyn_driver MoorDyn.dvr` in the command line. 

With regards to this error, I will defer to @sanguinariojoe who set up the wrapper. 

In the meantime, if you want to setup MoorDynC as a python module locally, you can do the following: 
1. Compile MoorDyn from the source code [here](https://moordyn.readthedocs.io/en/latest/compiling.html#compile-using-cmake)
2. Once that compiles successfully, change to `MoorDyn/build/wrappers/python/` and execute the following command `python3 setup.py install` This will build the python module locally from the source code you have installed.

tlopez-olocco, 2023-09-18:
Thanks @RyanDavies19 for your response, I generally use the OpenFAST precompiled binaries when I want to run a wind turbine simulation with moordyn, I could try to set up the MoorDyn driver. However, I would like to set up the MoorDynC version to couple with other codes I have.

I followed the instructions in the compiling section of moordyn.readthedocs.io. After dealing with some errors from my side I managed to get to the final step. After doing the mingw32-make in the C:\MoorDyn\MoorDyn.build directory, the following error appears:

LINK : warning LNK4044: unrecognized option '/lmoordyn'; ignored
   Creating library \MoorDyn_v2\\MoorDyn.build\\wrappers\\python\cmoordyn.cp311-win_amd64.lib and object \MoorDyn_v2\\MoorDyn.build\\wrappers\\python\cmoordyn.cp311-win_amd64.exp
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetRod
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetPoint
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Create
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetPointPos
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_ExternalWaveKinInit
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLineID
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_ExternalWaveKinSet
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetFASTtens
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Save
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetNumberLines
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetPointType
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetPointID
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_ExternalWaveKinGetN
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SaveVTK
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Init_NoIC
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetRodN
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetNumberBodies
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetNumberRods
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetBodyType
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetNumberPoints
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLineNumberNodes
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetBody
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SetLineUnstretchedLength
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SetVerbosity
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLineNodeTen
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Log
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Deserialize
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLine
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetPointForce
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SetLineUnstretchedLengthVel
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetWaves
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetWavesKin
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Step
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetPointAttached
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetPointVel
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Load
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLineFairTen
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Close
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLineMaxTen
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLineUnstretchedLength
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SaveRodVTK
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetBodyState
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SetLogLevel
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SaveBodyVTK
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Init
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_ExternalWaveKinGetCoordinates
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetRodID
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetPointNAttached
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLineNodePos
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLineNodeCurv
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetRodType
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_NCoupledDOF
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetBodyID
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_Serialize
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetRodNodePos
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SaveLineVTK
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SavePointVTK
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_GetLineN
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_SetLogFile
cmoordyn.obj : error LNK2001: unresolved external symbol MoorDyn_UseBodyVTK
[... cut ...]
After finishing this successfully, if I understood it well I could build the python module locally with python setup.py install. Nevertheless, just in case I executed the command just in case and this was the obtained message:

copying .\\moordyn\Generator\Rod.py -> build\lib.mingw_x86_64-cpython-310\moordyn\Generator
copying .\\moordyn\Generator\__init__.py -> build\lib.mingw_x86_64-cpython-310\moordyn\Generator
running build_ext
building 'cmoordyn' extension
error: --plat-name must be one of ('win32', 'win-amd64', 'win-arm32', 'win-arm64')

Can you think of any solution? What do you think could be the problem?

sanguinariojoe, 2023-09-18:
You cannot use setup.py on your own, you must let CMake do all the job. Otherwise you will have this kind of issues. You mentioned you had some problems on your side, can you elaborate?

I am fixing the sdist right now (see https://github.com/FloatingArrayDesign/MoorDyn/pull/130), so the installation with pip will work again. That would take some time though... It should be merged on dev, then on master and finally the autoproduced packages shall  be uploaded to pypi

tlopez-olocco, 2023-09-18:
Thanks for your answer @sanguinariojoe, 

One of the issues was that I didn't have pip upgraded up to date (so I had to update it in the msys2). Also, I had to install a newer version of Microsoft C++ Build Tools. After that, CMake compiled normally and created the necessary folders.

The problem came when I tried to execute mingw32-make.

Good to know what you fixed in #130. Still not working the straight forward installation with pip install moordyn though.

RyanDavies19, 2023-09-19:
@tlopez-olocco just to clarify, where are you in the compiling process and what error are you getting? Have you been able to build all the folders, including `wrappers/python/ ` with files in them?

tlopez-olocco, 2023-09-20:
@RyanDavies19 I followed the steps in https://moordyn.readthedocs.io/en/latest/compiling.html#compile-using-cmake

After enabling the MATLAB_WRAPPER and PYTHON_WRAPPER in CMake. I clicked “Configure” once more and then “Generate”. As I understand until this point everything was fine and the necessary folders were created. For example, the folder wrappers/python/:

[image]

The error comes when I try to do the following:

cd C:\MoorDyn_v2\MoorDyn.build
mingw32-make
mingw32-make install

sanguinariojoe, 2023-09-20:
The problem is that you are compiling with the MSVC compiler. see:

build\lib.win-amd64-cpython-311\cmoordyn.cp311-win_amd64.pyd : fatal error LNK1120: 60 unresolved externalserror: command '**C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.37.32822\bin\HostX86\x64\link.exe**' failed with exit code 1120

is that intentional? or you are doing that because of a mistake? (the solution depends on your answer, of course)

tlopez-olocco, 2023-09-20:
@sanguinariojoe, following the instructions of https://moordyn.readthedocs.io/en/latest/compiling.html#compile-using-cmake, I installed MSYS2 and typed the following command

pacman -S mingw-w64-x86_64-python-setuptools mingw-w64-x86_64-python-pip mingw64/mingw-w64-x86_64-make mingw-w64-x86_64-gcc mingw-w64-x86_64-gdb mingw-w64-x86_64-cmake

Then I made MinGW stack available across the whole system by adding it to the PATH environment variable. 

After that, I proceeded with the CMake building compiling process. Maybe I made a mistake that I am not aware of during this process.

Thanks.

sanguinariojoe, 2023-09-20:
Probably not your fault. Can you please remove MoorDyn.build folder and do the CMake process again? This time, on this window:

[image]

Select the "specify native compilers" option. I suppose (sorry, not a windiws guy) you will then be asked with a window like the following:

[image]

Just select the compilers you have C:\msys64\mingw64\bin (gcc, g++ and gfortran)

tlopez-olocco, 2023-09-21:
@sanguinariojoe As you suggested, I selected my native compilers and did the CMake process again:

[image]

[image]

If I uninstall the Visual Studio build tools the message I get is: error: ((( Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/ ))), instead of the ((( build\lib.win-amd64-cpython-311\cmoordyn.cp311-win_amd64.pyd : fatal error LNK1120: 60 unresolved externalserror: command 'C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.37.32822\bin\HostX86\x64\link.exe' failed with exit code 1120 )))

[image]

I am wondering, should I install more stuff that are missing like the following in my build tools?

[image]

[image]

[image]

Anyway, even if I install the VS build tools still nothing, the message I get is kind of the same:

C:\Users\tomil\AppData\Local\Temp\build-env-d337nyu0\Lib\site-packages\setuptools\config\expand.py:134: SetuptoolsWarning: File 'C:\\Users\\tomil\\AppData\\Local\\Temp\\build-via-sdist-6_53if4v\\moordyn-2.2\\README.md' cannot be found

[image]

[image]

I cannot think of anything else. I didn't find any hint online either.

Thanks.

sanguinariojoe, 2023-09-21:
Oh, I see, it is Python the one who is trying to compile with MSVC... Geez, Windows is a mess...

OK, let's forget about MinGW then... On this screen

[image]

Instead of "MinGW Makefiles" select the "Visual Studio blablabla".

That will create a Visual studio project for you on the MoorDyn.build folder. When you open that project you will notice a BUILD target as well as an INSTALL target. I recommend you to execute the BUILD target, then close Visual Studio and open it again with administrative rights to execute the INSTALL target

sanguinariojoe, 2023-09-28:
> ImportError: /home/tomas/Documents/moordyn/venv/lib/python3.10/site-packages/cmoordyn.cpython-310-x86_64-linux-gnu.so: undefined symbol: _ZN21vtkDataArraySelection3NewEv

Same problem here! I did not detect this problem before because I had VTK installed on the system. I have to fix this!

> Error using coupling_MD.m
> MoorDyn reported an error

MoorDyn is probably working, but your configuration file maybe not. We need more details

RyanDavies19, 2023-09-28:
@tlopez-olocco Just reaching out to let you know the python wrapper of MoorDyn has been updated to the newest version and can be found at [pypi](https://pypi.org/project/moordyn/). If you are still struggling to install MoorDyn, this would be an easy way to get up and running with the python module.

tlopez-olocco, 2023-10-03:
@RyanDavies19 Yes, indeed now I could install the latest version of moordyn with pip install. I am using pycharm now and works perfectly.

I think for windows, this is the easiest way for those who want to work with python.

Thanks!

PR #130 (2023-09-19): Make the sdist installable (Python)
Now the source distribution will be installable, without VTK support. I let the technical debt of autodownloading VTK static libraries

Fixes #122
closed 2023-10-03 by RyanDavies19