FloatingArrayDesign/MoorDyn #131: Unable to implement Compile using CMake

ldd289806329, 2023-09-19:
Hello Professors

I have a new question.
Firstly, I carefully and repeatedly followed your steps and requirements. Currently, I am able to complete the Configuration options for Eigen3. However, when I installed Eigen in C: Program Files (x86) Eigen3, I ran Command Prompt as the manager and typed the following [commands:](https://moordyn-v2.readthedocs.io/en/latest/compiling.html#:~:text=cd%20C%3A%5CMoorDyn%5CMoorDyn.build%0Amingw32%2Dmake%0Amingw32%2Dmake%20install)
<cd C:/MoorDyn/eigen.build
mingw32 make
mingw32 make install>
When I input<cd C: MoorDyn eigen. build>, it is still normal here, but there is no response when I input<mingw32 make>(because there is no such program or file under the eigen. build folder). When I run<mingw32 make install>, many. h files will be installed. I don't know what the function of type commands:<mingw32 make>(second line) is in the command prompt, or I don't know what I did wrong earlier. When I continue to follow your instruction file Configuration options for MoorDyn, an error will appear. The error is:
<The Fortran compiler identification is unknown
CMake Error at CMakeLists.txt:57 (enable_language):
No CMAKE_Fortran_COMPILER could be found.
Tell CMake where to find the compiler by setting either the environment
variable "FC" or the CMake cache entry CMAKE_Fortran_COMPILER to the full
path to the compiler, or to the compiler name if it is in the PATH.
Configuring incomplete, errors occurred!>
I don't know why, maybe it's because I'm a novice. I don't know how to find CMAKE_ Fortran_ COMPILER and address this issue. I request that you can create a teaching video about Compiling CMake and Compile as a Simple Library, and publish it online (such as YouTube). If you have time, you can do more application examples about MoorDyn, which will make it more convenient for scholars to use MoorDyn. As developers, you have also saved a lot of time answering questions for everyone. Because this instruction file is difficult for beginners to complete the installation. I hope to receive your reply, thank you!

sanguinariojoe, 2023-09-19:
> When I input, it is still normal here, but there is no response when I
> input(because there is no such program or file under the eigen. build
> folder). When I run, many. h files will be installed. I don't know what the
> function of type commands:(second line) is in the command prompt, or I

Are you interested on the Fortran wrapper? Otherwise just disable it on the
cmake options (right before clicking configure).

If you indeed want the fortran wrapper you can install the fortran compiler
on your msys2 (I do not know the package name, look for gfortran)

Anyway, Ryan and Matt we can simplify and improve the documentation
dropping the eigen stuff, as long as we provide an internal copy now. You
are the Windows guys, so please take lead on this.

In my experience youtube videos are usually created by users, not
developers.

There are plenty of applications on the tests folder.

MoorDyn is by definition a library to be coupled with other solvers, and
thus it can be hardly simplified for beginners. Anyway, we are investing
significant effort to have a Python package which will be easy to install
and use.

ldd289806329, 2023-09-19:
Sorry, after canceling Fortran, I did it again, but there are still error prompts. The error is:
[image]
It is different from your [document](https://moordyn-v2.readthedocs.io/en/latest/compiling.html#:~:text=Configuration%20options%20for%20MoorDyn).
If that, how can I solve it?
how can i install it typing  'C:/msys64/mingw64/bin/python3.10.exe -m pip install --upgrade build'
Thank you

sanguinariojoe, 2023-09-19:
Just do as it instruct you. Open a terminal and type that.

cc200528, 2023-09-19:
I had the same problem. I use windows, too. According to what you said, I opened theC:/msys64/mingw64/bin/python3.10.exe . What command should I type next?
[image]
Thank you.

RyanDavies19, 2023-09-19:
@ldd289806329 @cc200528 you should copy and paste the exact command in quotes that it says to use into your terminal. PyPa's is a python package that needs to be installed in order for MoorDyn to work. All of the errors here are coming from packages moordyn requires but you don't have installed.

RyanDavies19, 2023-09-19:
> Are you interested on the Fortran wrapper? Otherwise just disable it on the cmake options (right before clicking configure). If you indeed want the fortran wrapper you can install the fortran [...]

@sanguinariojoe I think it would be best to simplify, and keeping Eigen up to date can be something that happens on our end. I can try to work on changing the compile flags to account for that.

RyanDavies19, 2023-09-19:
@AlexWKinley do you have thoughts or suggestions about making the internal Eigen package the default and moving external Eigen to an optional compile flag?

RyanDavies19, 2023-09-19:
@AlexWKinley sounds good I will work on this. I've been using MoorDyn as a system library as a number of projects (MacOS) and it has worked fine with the internal Eigen.

sanguinariojoe, 2023-09-20:
The problem of the internal Eigen is that the final user cannot have access
to the C++ API, as long as any "eigen-ish" include will fail. On top of
that, we cannot install our own/internal copy of Eigen because it will
eventually collide with the official package.

Said that, C API should be enough for regular users. The C++ access is just
for really advanced ones.

sanguinariojoe, 2023-10-04:
Internal Eigen is the default option now. On top of that, we are now recommending either the pip package or the installer.

I hereby declare this issue solved

closed 2023-10-04 by sanguinariojoe