FloatingArrayDesign/MoorDyn #134: dev docs rendering

RyanDavies19, 2023-09-20:
Opening an issue so we can discuss rendering the dev docs on MoorDyn.readthedocs.io/en/latest. So far we have tried the following:

1. Specified sphinx_rtd_theme version per conversation in #129. This resolved the issue with bullet points not rendering however the max depth on the sidebar is now fixed at 2 and the menu to change builds in the bottom left corner is now broken. 

2. Changed toctree macdepth in index.rst to 5. This made no change. 

These issues don't seem to appear on MoorDyn-v2.readthedocs.io/en/latest, which is referencing the same docs folder in the dev branch. As far as we can tell this means there is an issue in how readthedocs is configured for MoorDyn.readthedocs.io/en/latest

sanguinariojoe, 2023-09-21:
My general settings are quite normal:

[image]

The advanced ones seem also rather normal:

[image]
[image]

RyanDavies19, 2023-09-21:
That PR did not fix things. Looking at the [difference in build logs](https://www.diffchecker.com/Ufuj8Fx4/) at line 170, moordyun.readthedocs is installing pillow instead of sphinx and then pillow is installing a depreciated version of sphinx. It's strange because the settings are identical to what you have @sanguinariojoe  and what I have on moordyn-ryan.readthedocs

RyanDavies19, 2023-09-21:
Created a dev_docs branch to test things on so that we don't fill up the dev branch with junk commits

RyanDavies19, 2023-09-21:
Nothing I have tried has worked so I gave up and submitted an issue at readthedocs: readthedocs/readthedocs.org#10765

RyanDavies19, 2023-09-25:
Resolved by deleting and remaking the project. @sanguinariojoe can we retire [moordyn-v2.readthedocs.io](https://moordyn-v2.readthedocs.io/en/latest/) now that [moordyn.readthedocs.io](https://moordyn.readthedocs.io/en/latest/) is built and properly formatted?

closed 2023-09-25 by RyanDavies19

sanguinariojoe, 2023-09-26:
https://moordyn-v2.readthedocs.io/en/latest/ is no more

PR #136 (2023-09-21): Fixed readthedocs (hopefully)
These changes fixed the readthedocs dev issues when I built my own version of the docs at MoorDyn-ryan.readthedocs.io. Once this is resolved I will delete the page. 

Hopefully resolves #134