FloatingArrayDesign/MoorDyn #135: Figuring out how to proceed on Windows

sanguinariojoe, 2023-09-21:
Windows is the worst, because is designed by the monkeys of the monkeys for the monkeys. Unfortunately we would deal with it, offering a Windows way of life for people wanting to use it.

Python solution is on its way:

https://github.com/FloatingArrayDesign/MoorDyn/pull/132

But on C we have some problems... Mainly because MinGW and MSVC are not compatible. Each one has its merits...

 - MinGW can easily deal with fortran, but it is rather impossible to build the Python wrapper on any modern CPython implementation, i.e. > 3.4 (see https://wiki.python.org/moin/WindowsCompilers#GCC_-_MinGW-w64_.28x86.2C_x64.29). Providing VTK support is also a pain in the ass.
 - MSVC needs a truckload of shit, and if you want the Fortran wrapper you need to add the Intel One SDK, which is another truckload of shit. On the other hand, it can easily deal with Python and VTK.

Thus, I suggest to change the documentation:

 - Recommend to Python users to install it with pip
 - Recommend to Windows C users to use the installer provided on the automatic releases (https://github.com/FloatingArrayDesign/MoorDyn/releases/download/nightly/Moordyn-2.2.0-win64.exe was the last one)
 - Provide a compilation guide for both MSVC and MinGW for more advanced users (if they want to deal with that)

Can any of you, @RyanDavies19 or @AlexWKinley, try out the [Windows installer](https://github.com/FloatingArrayDesign/MoorDyn/releases/download/nightly/Moordyn-2.2.0-win64.exe)? it is a bit outdated, but shall be functional anyway

RyanDavies19, 2023-09-21:
@sanguinariojoe I am on MacOS so I can't. However someone from the WECSim team has compiled for windows (with the Matlab wrapper). He said it worked fine once he disabled the Fortran and Python wrappers.

AlexWKinley, 2023-09-21:
Trying out the windows installer I've thus far run into 2 strange issues. 

Firstly, during the install process, when I ask it to add MoorDyn to the system path the installer throws an error that the system path is too long, and it can't add to it.
[image]

I'm certainly able to add to the path myself, so I'm not sure what's the cause of this error.

The next issue I ran into when trying to use the installed MoorDyn was that it wanted the moordynf.lib file but it didn't exist.

My cmake file looked like
```
cmake_minimum_required(VERSION 3.10)

project(MDTest)

add_executable(MDTest
    src/main.cpp
)

find_package(MoorDyn REQUIRED)
target_link_libraries(MDTest
    MoorDyn::moodyn
)
```

And the output of running cmake (through vscode) is 
```
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -SC:/Users/A.Kinley/Documents/moordyn_install_test -Bc:/Users/A.Kinley/Documents/moordyn_install_test/build -G "Visual Studio 17 2022" -T host=x64 -A x64
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
[cmake] CMake Error at C:/Program Files/MoorDyn-2.2/lib/cmake/moordyn/MoorDynTargets.cmake:98 (message):
[cmake]   The imported target "MoorDyn::moordynf" references the file
[cmake] 
[cmake]      "C:/Program Files/MoorDyn-2.2/lib/moordynf.lib"
[cmake] 
[cmake]   but this file does not exist.  Possible reasons include:
[cmake] 
[cmake]   * The file was deleted, renamed, or moved to another location.
[cmake] 
[cmake]   * An install or uninstall procedure did not complete successfully.
[cmake] 
[cmake]   * The installation package was faulty and contained
[cmake] 
[cmake]      "C:/Program Files/MoorDyn-2.2/lib/cmake/moordyn/MoorDynTargets.cmake"
[cmake] 
[cmake]   but not all the files it references.
[cmake] 
[cmake] Call Stack (most recent call first):
[cmake]   C:/Program Files/MoorDyn-2.2/lib/cmake/moordyn/MoorDynConfig.cmake:19 (include)
[cmake]   CMakeLists.txt:10 (find_package)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[proc] The command: "C:\Program Files\CMake\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -SC:/Users/A.Kinley/Documents/moordyn_install_test -Bc:/Users/A.Kinley/Documents/moordyn_install_test/build -G "Visual Studio 17 2022" -T host=x64 -A x64 exited with code: 1
```

I'm not familiar enough with how things are set up and the like to know if this is an error of my own causing, or if a file isn't getting included that should be. 

Keep me in the loop if there's anything I can do to help with windows stuff, Kelson primarily uses Windows so I have access to both my development machine and other machines that are yet untouched by MoorDyn.

sanguinariojoe, 2023-09-22:
Argh! Sorry! Totally my bad... the installer (master branch) is way more
outdated than i thought. It is still using the intel one sdk.

I will try to craft you an installer later.

Ty so much!

sanguinariojoe, 2023-09-22:
@AlexWKinley you have the new installer [here](https://github.com/core-marine-dev/MoorDyn/releases/tag/v2.2.1). Please, give it a shot

AlexWKinley, 2023-09-22:
@sanguinariojoe Well besides the same path error problem, it unfortunately still doesn't seem to work, but for a different reason. 

I used the same cmake file (although fixed the typo in target link libraries), and now it succesfully builds, but when trying to run, it tries to find a moordyn.dll, fails, and then exits. It seems like the installer provided a moordyn.lib, which somehow didn't get statically linked I guess? If I look at the executable dependencies, it lists moordyn.dll, and that it didn't find it
[image]

I'll attach the entire test project folder, including the build folder, in case that provides any useful information. 
I did compile with clang on windows (using the windows toolchain), but compiling with MSVC ran into the same issue, and I figured the ninja files, and clang commands might be a little easier to look through than solution files.

We're in agreement on the frustration of windows. As always let me know what assistance I can provide. 
[moordyn_install_test.zip](https://github.com/FloatingArrayDesign/MoorDyn/files/12702210/moordyn_install_test.zip)

sanguinariojoe, 2023-09-26:
I tried [this](https://github.com/core-marine-dev/MoorDyn/releases/tag/v2.2.2) and works just fine with your test @AlexWKinley . Regarding the PATH issues, it is caused by your own PATH which is already too long. In my testing computer it worked

AlexWKinley, 2023-09-26:
@sanguinariojoe It looks like once I fix my path to have `C:\Program Files\MoorDyn-2.2\bin`, then worked with that build you linked. I would agree my path is probably longer than most users'. So I'm in agreement that things work.

sanguinariojoe, 2023-10-04:
This is then fixed on #139

PR #139 (2023-10-04): Fixed the pre-compiled binaries, tested on Windows
Fixes #135
closed 2023-10-04 by sanguinariojoe