FloatingArrayDesign/MoorDyn #142: Default argument error with C API: Waves.h

RyanDavies19, 2023-09-27:
When using load library in Matlab to load the C API, the following error message is produced: 

```
------Error in the Model, Error-Causes/Messsages are :
Building libmoordyn_thunk_maci64 failed.  Compiler output is:
/usr/bin/xcrun -sdk macosx14.0 clang -I"/Applications/MATLAB_R2023a.app/extern/include" -fno-common -arch x86_64 -mmacosx-version-min=10.15 -fexceptions -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk -I"/Users/rdavies/work/wecsim_libs/WEC-Sim_Applications/Mooring/MoorDyn" -I"/Users/rdavies/work/wecsim_libs/WEC-Sim/source/functions/moorDyn" "libmoordyn_thunk_maci64.c" -o "libmoordyn_thunk_maci64.dylib" -bundle 
In file included from libmoordyn_thunk_maci64.c:27:
In file included from /Users/rdavies/work/wecsim_libs/WEC-Sim/source/functions/moorDyn/MoorDyn2.h:39:
/Users/rdavies/work/wecsim_libs/WEC-Sim/source/functions/moorDyn/Waves.h:77:59: error: C does not support default arguments
                                        MoorDynSeafloor seafloor = 0);
```

I worked around it by removing the seafloor input at line 76 in Waves.h and at line 899 in Waves.cpp. At line 909 I replaces the seafloor variable with 0. This a temporary fix because it removes the capability, but necessary for use of the C API. 

Note: we are not using the Matlab wrapper with WECSim because of compatibility issues with older versions of Matlab

sanguinariojoe, 2023-09-28:
Yeah, @AlexWKinley we should be careful of not adding C++ stuff on the .h files.

We are fixing this for 2.2.2

sanguinariojoe, 2023-10-09:
Since this require changing the ABI, I decided to move this to the next minor release, 2.3.0

commit 5d57d463 (2023-12-22): Fix #142
closed 2024-05-23 by RyanDavies19