FloatingArrayDesign/MoorDyn #144: Inputs of MoorDyn_Step function

soydanahmet, 2023-10-02:
Hi,

I am currently in the process of coupling Mordyn v2.2.1 with my code, and I am seeking clarification regarding the inputs of the MoorDyn_Step function, specifically in terms of their definitions and units. Your assistance in clarifying this matter would be greatly appreciated.

The MoorDyn_Step function has the following input parameters:

MoorDyn_Step(MoorDyn system, const double *x, const double *xd, double *f, double *t, double *dt)

Based on my understanding, the input vectors x (Position vector) and xd (Velocity vector) should be structured as follows:

x : {CoM_X, CoM_Y, CoM_Z, phi, theta, psi}

CoM_X: Coordinate the center of mass in the x-direction (meter)
CoM_Y: Coordinate the center of mass in the y-direction (meter)
CoM_Z: Coordinate the center of mass in the z-direction (meter)

**phi**:  roll angle (rad)
**theta**: pitch angle (rad)
**psi**: yaw angle (rad)

**xd** : {V_X, V_Y, V_Z, omega_I(0), omega_I(2), omega_I(2)}

V_X: Velocity of the center of mass in the x-direction (m/s)
V_Y: Velocity of the center of mass in the y-direction (m/s)
V_Z: Velocity of the center of mass in the z-direction (m/s)

omega_I(0): Angular velocity around the x-direction (rad/s)
omega_I(1): Angular velocity around the y-direction (rad/s)
omega_I(1): Angular velocity around the z-direction (rad/s)

Thanks in advance.

soydanahmet, 2023-10-03:
Thank you so much for your prompt response. Now my code is working. I have one more (and last :) question about the moorDyn time step.

I am currently fine-tuning its integration with my computational framework. However, I might encounter a situation where the time step size in my code, which is controlled under the CFL condition, occasionally results in very small time step sizes. In contrast, the MoorDyn time step, as initially defined in the input file, may remain relatively large. So I'm curious whether it is possible to adjust the MoorDyn time step dynamically during the simulation.

sanguinariojoe, 2023-10-03:
If you ask MoorDyn to integrate a time step, dt, which is smaller than the
inner/model time step, dtM, then dtM is automagically corrected as dtM = dt

Even more, dt will be always honoured. If you ask to integrate a dt which
is not divisible by dtM, then Moordyn will do as many steps with dtM as
possible followed by a last smaller one.

closed 2023-10-03 by soydanahmet