FloatingArrayDesign/MoorDyn #146: Runtime issue using CMake

RyanDavies19, 2023-10-02:
I'm not sure at what point things were changed, but MD-C compiled with CMake is now significantly slower compared to MD-C compiled with the makefiles in the `compile/` folder. For the case1 attached file, the table below shows the runtimes from my Mac when simulated for 50 seconds with a static fairlead:

MD-F:
- real    0m3.390s
- user    0m3.216s
- sys     0m0.541s
MD-C compile/DYLIB 
- real    0m2.168s
- user    0m2.337s
- sys     0m0.467s
MD-C build/ 
- real    1m36.923s
 - user    1m36.363s
 - sys     0m0.583s

[case1.txt](https://github.com/FloatingArrayDesign/MoorDyn/files/12786939/case1.txt)

sanguinariojoe, 2023-10-03:
-DCMAKE_BUILD_TYPE=Release ?

Another question would be, Do we want to make Release the default type?

sanguinariojoe, 2023-10-04:
> @sanguinariojoe <https://github.com/sanguinariojoe>
> I think I generally agree with your sentiment, debug as default is the
> most common behavior among all build systems.
> That being said Eigen is somewhat notorious for producing slow debug
> builds because of its template wizardry, and so as a result the debug
> builds are slow to the point of not being usable.
>
> I think my vote would be for RelWithDebInfo being the default, since you
> get basically all of the speed benefits and still maintain some debug info.
> And given that MoorDyn is generally likely to be used by people who are not
> developers, the more usable defaults there are, the easier it will be.

I am pretty sure relwithdebuginfo is the default option. Would you check
that?

RyanDavies19, 2023-10-04:
@AlexWKinley @sanguinariojoe I will defer to you all on what is the best option because I am not super familiar with cmake build options. I will echo Alex's note about making it as user friendly as possible, because one of the biggest barriers to use right now is most users are not developers.

AlexWKinley, 2023-10-04:
@sanguinariojoe 
At least on my system it seems to default to debug. Running
```
cmake .. -GNinja -DEXTERNAL_EIGEN="OFF" -DPYTHON_WRAPPER="OFF"
```
from an empty build folder in master uses a debug build, putting this in the CmakeCache.txt
```
//Choose the type of build, options are: None Debug Release RelWithDebInfo
// MinSizeRel ...
CMAKE_BUILD_TYPE:STRING=Debug
```
And it's mirrored in the build.ninja
```
build source/CMakeFiles/moordyn.dir/Body.cpp.obj: CXX_COMPILER__moordyn_unscanned_Debug C$:/Users/A.Kinley/Programs/MoorDyn/source/Body.cpp || cmake_object_order_depends_target_moordyn
  DEFINES = -DMOORDYN_MAJOR_VERSION=2 -DMOORDYN_MINOR_VERSION=2 -DMoorDyn_EXPORTS -Dmoordyn_EXPORTS
  DEP_FILE = source\CMakeFiles\moordyn.dir\Body.cpp.obj.d
  FLAGS = -O0 -g -Xclang -gcodeview -D_DEBUG -D_DLL -D_MT -Xclang --dependent-lib=msvcrtd -std=gnu++17
  INCLUDES = -IC:/Users/A.Kinley/Programs/MoorDyn/build2/source -IC:/Users/A.Kinley/Programs/MoorDyn/source
  OBJECT_DIR = source\CMakeFiles\moordyn.dir
  OBJECT_FILE_DIR = source\CMakeFiles\moordyn.dir
  TARGET_COMPILE_PDB = source\CMakeFiles\moordyn.dir\
  TARGET_PDB = source\moordyn.pdb
  ```

sanguinariojoe, 2023-10-04:
Interesting... I would say the makefiles generator (xmake is called?)
behaves differently.

Anyway, I will check the relwithdebuginfo performance and if it is good
enough i will set it as default.

Ryan, now we have the python packages and the installers. I do not think
CMake will be a barrier on the future.

sanguinariojoe, 2023-10-09:
**None:**

Neverending history...

**Debug:**

Neverending history...

**Release:**

```
      Start  1: minimal
 1/18 Test  #1: minimal ..........................   Passed    1.09 sec
      Start  2: lifting
 2/18 Test  #2: lifting ..........................   Passed    1.59 sec
      Start  3: pendulum
 3/18 Test  #3: pendulum .........................   Passed   18.88 sec
      Start  4: beam
 4/18 Test  #4: beam .............................   Passed   33.66 sec
      Start  5: time_schemes
 5/18 Test  #5: time_schemes .....................   Passed   27.78 sec
      Start  6: io
 6/18 Test  #6: io ...............................   Passed    1.04 sec
      Start  7: rods
 7/18 Test  #7: rods .............................   Passed    4.04 sec
      Start  8: bodies_and_rods
 8/18 Test  #8: bodies_and_rods ..................   Passed    1.93 sec
      Start  9: quasi_static_chain
 9/18 Test  #9: quasi_static_chain ...............   Passed   16.84 sec
      Start 10: wavekin_4
10/18 Test #10: wavekin_4 ........................   Passed    2.19 sec
      Start 11: tripod
11/18 Test #11: tripod ...........................   Passed    5.29 sec
      Start 12: seafloor
12/18 Test #12: seafloor .........................   Passed    6.92 sec
      Start 13: wavekin
13/18 Test #13: wavekin ..........................   Passed   21.78 sec
      Start 14: wavekin_7
14/18 Test #14: wavekin_7 ........................   Passed    4.51 sec
      Start 15: bodies
15/18 Test #15: bodies ...........................   Passed    0.19 sec
      Start 16: math_tests
16/18 Test #16: math_tests .......................   Passed    0.00 sec
      Start 17: testDecomposeString
17/18 Test #17: testDecomposeString ..............   Passed    0.00 sec
      Start 18: fortran_bindings
18/18 Test #18: fortran_bindings .................   Passed    0.44 sec

100% tests passed, 0 tests failed out of 18

Total Test time (real) = 148.19 sec
```

**RelWithDebInfo:**

```
      Start  1: minimal
 1/18 Test  #1: minimal ..........................   Passed    1.27 sec
      Start  2: lifting
 2/18 Test  #2: lifting ..........................   Passed    1.77 sec
      Start  3: pendulum
 3/18 Test  #3: pendulum .........................   Passed   17.57 sec
      Start  4: beam
 4/18 Test  #4: beam .............................   Passed   37.51 sec
      Start  5: time_schemes
 5/18 Test  #5: time_schemes .....................   Passed   30.49 sec
      Start  6: io
 6/18 Test  #6: io ...............................   Passed    1.14 sec
      Start  7: rods
 7/18 Test  #7: rods .............................   Passed    5.12 sec
      Start  8: bodies_and_rods
 8/18 Test  #8: bodies_and_rods ..................   Passed    2.28 sec
      Start  9: quasi_static_chain
 9/18 Test  #9: quasi_static_chain ...............   Passed   19.79 sec
      Start 10: wavekin_4
10/18 Test #10: wavekin_4 ........................   Passed    2.41 sec
      Start 11: tripod
11/18 Test #11: tripod ...........................   Passed    6.15 sec
      Start 12: seafloor
12/18 Test #12: seafloor .........................   Passed    6.59 sec
      Start 13: wavekin
13/18 Test #13: wavekin ..........................   Passed   25.67 sec
      Start 14: wavekin_7
14/18 Test #14: wavekin_7 ........................   Passed    4.76 sec
      Start 15: bodies
15/18 Test #15: bodies ...........................   Passed    0.21 sec
      Start 16: math_tests
16/18 Test #16: math_tests .......................   Passed    0.00 sec
      Start 17: testDecomposeString
17/18 Test #17: testDecomposeString ..............   Passed    0.00 sec
      Start 18: fortran_bindings
18/18 Test #18: fortran_bindings .................   Passed    0.49 sec

100% tests passed, 0 tests failed out of 18

Total Test time (real) = 163.23 sec
```

**MinSizeRel:**

```
      Start  1: minimal
 1/18 Test  #1: minimal ..........................   Passed    2.13 sec
      Start  2: lifting
 2/18 Test  #2: lifting ..........................   Passed    3.12 sec
      Start  3: pendulum
 3/18 Test  #3: pendulum .........................   Passed   37.37 sec
      Start  4: beam
 4/18 Test  #4: beam .............................   Passed   80.02 sec
      Start  5: time_schemes
 5/18 Test  #5: time_schemes .....................   Passed   56.94 sec
      Start  6: io
 6/18 Test  #6: io ...............................   Passed    2.89 sec
      Start  7: rods
 7/18 Test  #7: rods .............................   Passed   15.60 sec
      Start  8: bodies_and_rods
 8/18 Test  #8: bodies_and_rods ..................   Passed    4.15 sec
      Start  9: quasi_static_chain
 9/18 Test  #9: quasi_static_chain ...............   Passed   36.50 sec
      Start 10: wavekin_4
10/18 Test #10: wavekin_4 ........................   Passed    5.49 sec
      Start 11: tripod
11/18 Test #11: tripod ...........................   Passed   10.92 sec
      Start 12: seafloor
12/18 Test #12: seafloor .........................   Passed   12.51 sec
      Start 13: wavekin
13/18 Test #13: wavekin ..........................   Passed   47.85 sec
      Start 14: wavekin_7
14/18 Test #14: wavekin_7 ........................   Passed    8.85 sec
      Start 15: bodies
15/18 Test #15: bodies ...........................   Passed    0.45 sec
      Start 16: math_tests
16/18 Test #16: math_tests .......................   Passed    0.00 sec
      Start 17: testDecomposeString
17/18 Test #17: testDecomposeString ..............   Passed    0.00 sec
      Start 18: fortran_bindings
18/18 Test #18: fortran_bindings .................   Passed    0.90 sec

100% tests passed, 0 tests failed out of 18

Total Test time (real) = 325.72 sec
```

PR #149 (2023-10-10): Cmake changes
Fixes #146 and set the Catch config variables as advanced
closed 2023-10-10 by sanguinariojoe