FloatingArrayDesign/MoorDyn #148: Euler convention for 6DOF kinematics

guillencampa, 2023-10-05:
Dear all,

I have a question related to the euler convention followed in MoorDyn v2.

As stated in read the docs the iposition for a coupled body should be [x, y, z, roll, pitch, yaw]. It is also stated that the input and output for rotations in MoorDyn-C use Euler angles.

Which euler convention is employed? Or more specifically, which order is employed to multiply the classic rotation matrices?

Thank you very much. Best regards, 

Guillén

RyanDavies19, 2023-10-05:
Hi @guillencampa, if you are using MD-C then internally MoorDyn uses quaternions to handle to the orientation and rotation of 6DOF objects. The input output Euler angles describe the roll pitch and yaw of the objects, in other words the input/outputs are the rotations about the X(roll), Y(pitch), and Z(yaw) axes of the reference frame.

https://moordyn.readthedocs.io/en/latest/theory.html#orientation-of-6-dof-objects

guillencampa, 2023-10-05:
Thank you for your response @RyanDavies19. But what is the order in which this rotations are applied? The same quaternion or rotation matrix will result in different rotation angles depending on the order/convention considered. As far as I know, usually roll, pitch and yaw angles describe de ZYX convention, rotating as Z -> Y -> X. Is this the case in MD-C?

AlexWKinley, 2023-10-05:
I believe the rotation order is using the XYZ convention. 
The code that does the conversion is in source/Misc.hpp
```c++
inline quaternion
Euler2Quat(const vec3& angles)
{
	using AngleAxis = Eigen::AngleAxis<real>;
	quaternion q = AngleAxis(angles.x(), vec3::UnitX()) *
	               AngleAxis(angles.y(), vec3::UnitY()) *
	               AngleAxis(angles.z(), vec3::UnitZ());
	return q;
}
```
If you want to test the results of this function you can use this link
https://godbolt.org/z/3Y73PjcYh
to see the generated quaternion and rotation matrix for a given roll, pitch, and yaw.

guillencampa, 2023-10-06:
Thank you both for your responses, @AlexWKinley  @RyanDavies19. I have checked and as you said, the rotation order uses XYZ convention.

closed 2023-10-06 by guillencampa