FloatingArrayDesign/MoorDyn #160: Static line tension disagreement

RyanDavies19, 2023-10-12:
Static lines initialize with different tensions between MDC_v2 and MDF_v2/MDC_v1. Differences also emerge with dynamic relaxation. Using [MoorPy](https://github.com/NREL/MoorPy), NREL's quasi-static model, the equilibrium fairlead tension is 4808 kN. This indicates that MDC_v1 and MDF_v2 are getting a solution closer to the correct answer. See the figures below:

 Line profile:
[image]

Fairlead tension with TmaxIC = 0:
[image]

Fairlead tension with TmaxIC = 200:
[image]

RyanDavies19, 2023-10-12:
@AlexWKinley Sure thing, here it is: 
[case1.txt](https://github.com/FloatingArrayDesign/MoorDyn/files/12886501/case1.txt)

RyanDavies19, 2023-10-12:
I've noticed dynamic relaxation differences popping up elsewhere too. I am currently working on tracking them down to put together a fix. @AlexWKinley if you see something that pops out to you that would be great to know.

sanguinariojoe, 2023-10-12:
4808 x 1.025 = 4928

Might it be just a different default water density?

RyanDavies19, 2023-10-12:
@sanguinariojoe I don't believe so, at least not from the default value point. For MDC v2: https://github.com/FloatingArrayDesign/MoorDyn/blob/99cd2b6e6afa7e5d58e2ac7ac3a4b05e127f1f72/source/MoorDyn2.cpp#L127

For MDC v1: https://github.com/FloatingArrayDesign/MoorDyn/blob/ea5ccf8c050f359cdd13a0a8b6f5cc5f3158405c/MoorDyn.cpp#L307

And MDF requires a manual input of rho via a driver input file, it is set to 1025 there.

RyanDavies19, 2023-10-12:
I do believe though it is a default value or scale issue of some kind because of the constant difference

sanguinariojoe, 2023-10-13:
It was just an idea. I let you guys deal with this. I have some other
projects to attend

AlexWKinley, 2023-10-13:
@RyanDavies19 
One thing that seems to have an effect is the fact that MoorDyn v2 does submergence calculations that make a decent approximation for cylinders. 

This causes the top segment to have slightly less buoyancy because it's like top corner is sticking out of the water.
Crude illustration of what part of the cylinder is out of the water
[image]

 If I remove the partial submergence calculation in Line::getStateDeriv
```c++
// F[i] = calcSubSeg(i, i + 1, surface_height);
F[i] = 1.0;
```
the fairlead tension does decrease to be a little closer to the others.
[image]

But it doesn't look like that is the only factor. You might want to validate my findings with your setup, I'm using a slightly old and slightly modified version of MoorDyn v2 (although my numbers generally agree with yours before the submergence change).

RyanDavies19, 2023-10-13:
@AlexWKinley I think it was the first one you noted, with regards to adding an extra weight. V1 did have the added weight found here: https://github.com/FloatingArrayDesign/MoorDyn/blob/ea5ccf8c050f359cdd13a0a8b6f5cc5f3158405c/Line.cpp#L324
Removing the added weight in v2 causes all the versions to line up (under the TmaxIC = 0 case).  Here is the comparison with that weight removed:
[image]

Another note though is that it seems like the dynamic relaxation methods have some differences. I will leave this issue open until we can resolve those as well. With the removal of that added weight, here are the fairtens with TmaxIC = 200:
[image]

@mattEhall do you have any inputs on either of these?

mattEhall, 2023-10-13:
Hi all,
Yeah, I agree this line https://github.com/FloatingArrayDesign/MoorDyn/blob/d1010520a4285f650c696a0ba0545c3ce58fe756/source/Line.hpp#L434 is causing a double counting of the weight contribution to the tension of line end nodes. If we remove it the code will be consistent with v2 Fortran.

Regarding differences in the initial conditions between C and F, the first thing I'd wonder is if the codes hit the convergence tolerance at different times (assuming they don't go through all of TMaxIC=200 seconds). I don't recall if the convergence criterion is the same in both codes.

RyanDavies19, 2023-10-13:
@mattEhall That's likely it, I believe the Fortran code reaches it first. I am working on that now and will have a PR up soon with both the weight double counting fix and that relaxation criteria standardized.

RyanDavies19, 2023-10-17:
Okay so it turned out there were a couple things going on here. First, MDF had an indexing error that caused it to not compare the current fairten to the previous 10 in the dynamic relaxation routine. Secondly MDC was checking the previous 10 fairtens while MDF was checking the previous 9 fairtens. Lastly the way the MDC main IC gen time stepping loop was set up caused an extra timestep to be run because ICTmax - t > 0 when t = ICTmax due to precision errors. Fixing these in both codes resolves the differences we were seeing. For that same case1 input file these are the results now: 

ICTmax = 0:
[image]

ICTmax = 200:
[image]

I am putting together PR's to both MDF and MDC with these changes.

PR #164 (2023-10-19): Adjustments to Dynamic relaxation
Resolves #160 

This PR removes the double counting of line weight and changes the dynamic relaxation routine to match the now fixed fortran code. The changes to the dynamic relaxation routine are as follows:
- The user inputted dtIC is subdivided into 10 steps, with the intent that the dtIC is the time that the user wants to see stability over for convergence
- Changes number of convergence iterations so that 9 iterations are compared to the current fairten (because of subdividing each dtIC into 10 pieces)
- Changes the while loop parameter to a tolerance (th [... cut]
closed 2023-10-17 by RyanDavies19