FloatingArrayDesign/MoorDyn #168: MoorDyn step calculate

Echo-new, 2023-11-07:
Dear sir
I want to know that when we calculate three mooring lines, we actually calculate each of them separately or in coupling?

sanguinariojoe, 2023-11-07:
I do not quite understand. You can solve as many lines as you want. Whether they are coupled or not depends on the system you setup. Have you took a look on [the tests](https://github.com/FloatingArrayDesign/MoorDyn/tree/master/tests)?

Please, elaborate a bit your question

Echo-new, 2023-11-09:
Dear sir
Thank you for your reply. I mean that when I separately set the number of mooring line to one and three, with the same line properties, the same motion of fairleads, it gives the same displacement and velocity data (Line1). 
Line1(12)
1        fixed    853.87     0.0    -320.0       0        0        0        0   
2        vessel     5.2      0.0     -70.0    0     0      0     0

Line1(14)、Line2(25)、Line3(36)
1        fixed    853.87     0.0    -320.0       0        0        0        0   
2        fixed   -426.94   739.47   -320.0       0     0      0     0
3        fixed   -426.94  -739.47   -320.0      0     0      0     0
4        vessel     5.2      0.0     -70.0      0     0      0     0
5        vessel    -2.6      4.5     -70.0      0     0      0     0
6        vessel    -2.6     -4.5     -70.0      0     0      0     0
But, they obviously belong to different mooring systems. And in the source code line.cpp, the displacement—r is a vector of N*1. So I would like to know that they are same by accident, or we just calculate each mooring line separately.

sanguinariojoe, 2023-11-09:
I assume you are using Moordyn v2. Otherwise please upgrade.

If you look to the initialization documentation:

https://moordyn.readthedocs.io/en/latest/drivers.html#group__new__c__api_1ga80fe35c444bfb979b45d627610dd4de4

It says "(6 components per coupled body or cantilevered rod and 3
components per pinned rod or coupled point)". So in your case (3 fairleads)
the displacement r shall be of lengh 3x3=9

That is the case indeed on the minimal test:

https://github.com/FloatingArrayDesign/MoorDyn/blob/master/tests/minimal.cpp#L114

which is considering 3 lines:

https://github.com/FloatingArrayDesign/MoorDyn/blob/master/tests/Mooring/lines.txt#L19

Echo-new, 2023-11-09:
Thank you, sir. I made the mistake of setting up only one mooring line. I just want to make a comparison. And I know what you mean, I know that x is a vector of 9*1, which is the displacement of the fairleads. But I actually want to know in line.cpp, each line node position-r is a vector of (N+1)*1, whether we calculate each line separately.
[https://github.com/FloatingArrayDesign/MoorDyn/blob/master/source/Line.cpp#L182](url)

RyanDavies19, 2023-11-09:
To clarify, each line in your system will have its own r vector created. And each r vector is really a vector of node position vectors with the structure r = [node number][node x/y/z]. If you want MoorDYn to output specific node positions, you can do that with the following output: LINEnNmPX where n is the line number and m is the node number. You can read more here: [https://moordyn.readthedocs.io/en/latest/inputs.html#id5](https://moordyn.readthedocs.io/en/latest/inputs.html#id5)

Echo-new, 2023-11-10:
I got it, thank you.

closed 2023-11-10 by Echo-new