FloatingArrayDesign/MoorDyn #183: Nonlinear Stiffness of a line

waiyanthetpaing, 2024-01-15:
Hi,

I am coupling with MoorDynV2 and openfoam using foamMooring.  However, I cannot define nonlinear stiffness of a line.  Could you please provide a format and required fields?

[stiffness.txt](https://github.com/FloatingArrayDesign/MoorDyn/files/13934285/stiffness.txt)
[image]

RyanDavies19, 2024-01-15:
Hi @waiyanthetpaing. Are you using the dev or the main branch of MoorDyn? This looks like a bug that has been recently resolved on the dev branch but was not included in the most recent releas.

waiyanthetpaing, 2024-01-15:
Hi @RyanDavies19 , thank you for the reply.  I am just using MoorDyn to couple with OpenFOAM from this link https://gitlab.com/hfchen20/foamMooring.  I am afraid that I have no idea what version it is using.

sanguinariojoe, 2024-01-15:
master then.

You can either change the branch yourself or wait until the next release

To change the branch just edit the .gitsubmodules file on foamMooring root
folder

hfchen20, 2024-01-18:
Switching to dev branch resolved this issue, as reported by the user at [foamMooring](https://gitlab.com/hfchen20/foamMooring/-/issues/5#note_1729890106).

RyanDavies19, 2024-01-18:
Good to hear. Closing this issue.

closed 2024-01-18 by RyanDavies19

sanguinariojoe, 2024-01-19:
P.S. I definitely shall ask Hai Fei to subscribe me to foamMooring issues