FloatingArrayDesign/MoorDyn #220: Double counting of added mass in rods and lines

RyanDavies19, 2024-06-13:
I was looking through the code in writing the theory and it appears we are double counting the added mass for lines and rods. We include the added mass in the mass matrix calculations, but then also include a fluid inertia force: 

Rod mass: https://github.com/FloatingArrayDesign/MoorDyn/blob/0cfd542d4c146b4e109e5676e712270121ae6206/source/Rod.cpp#L1040
Rod force (Aq and Ap): https://github.com/FloatingArrayDesign/MoorDyn/blob/0cfd542d4c146b4e109e5676e712270121ae6206/source/Rod.cpp#L1189

Line mass: https://github.com/FloatingArrayDesign/MoorDyn/blob/0cfd542d4c146b4e109e5676e712270121ae6206/source/Line.cpp#L889
Line force (Aq and Ap): https://github.com/FloatingArrayDesign/MoorDyn/blob/0cfd542d4c146b4e109e5676e712270121ae6206/source/Line.cpp#L1214

This fluid inertia force is not included in MD-F

AlexWKinley, 2024-06-13:
I believe that this is actually correct behavior. It's been a while since I've derived these equations, but the important distinction between the two is that the mass matrix is used to account for the added mass due to acceleration of the object itself, and the fluid inertia force is due to the acceleration of the fluid in a global coordinate system.

From the wikipedia page for the Morison equation for a moving body in an oscillatory flow
$$F = \rho V \dot{u} + \rho C_a V (\dot{u} - \dot{v}) + \frac{1}{2} \rho C_d A (u - v) |u - v|$$

The last term is the drag term, and the first two terms are the froude-krylov force and the hydrodynamic mass force respectively. 

With some algebra (and ignoring the drag term)
$$F = \rho V (C_a + 1) \dot{u}  - \rho V C_a \dot{v} $$
Where $C_a + 1$ is the inertia coefficient. And if you work through the algebra with the $\dot{v}$ acceleration term, and newtons second law $F = m a$, you end up with two fluid inertia based terms. One dependent on the fluid acceleration in the global frame, and one that effectively adds mass to the object.

RyanDavies19, 2024-06-13:
@AlexWKinley thank you for the clarification on this, that makes a lot more sense. And in looking at MD-F theres a comment saying that term has yet to be added. Closing this.

closed 2024-06-13 by RyanDavies19