FloatingArrayDesign/MoorDyn #246: when I build MoorDyn in OpenFOAM by foamMooring ,which show this error and can't use in OpenFOAM

byChen47, 2024-08-16:
when I build MoorDyn by foamMooring ,which show this error :CMake Warning at CMakeLists.txt:146 (message):valgrind is not found, you will not be able to run memchecks  ,and after I complete the MoorDyn of the foamMooring ,and test use foamMooring/tutorial/sixDoF_2D/deformMesh/,this MoorDyn show a error ,and can't use MoorDyn , So I want to ask it must need valgrind when build MoorDyn envirenment.
[image]
[image]

byChen47, 2024-08-16:
this is the message of MoorDyn error
[interFoam.txt](https://github.com/user-attachments/files/16634875/interFoam.txt)

sanguinariojoe, 2024-08-16:
It is not an error, it is a warning. You do not need valgrind.

MoorDyn is clearly working, as you can see on the following lines:

```
#3  moordyn::StationaryScheme::Step(double&) in ~/OpenFOAM/chen-v2206/platforms/linux64GccDPInt32Opt/lib/libmoordyn.so.2
#4  moordyn::MoorDyn::icStationary() in ~/OpenFOAM/chen-v2206/platforms/linux64GccDPInt32Opt/lib/libmoordyn.so.2
#5  moordyn::MoorDyn::Init(double const*, double const*, bool) in ~/OpenFOAM/chen-v2206/platforms/linux64GccDPInt32Opt/lib/libmoordyn.so.2
```

It really seems that your MoorDyn config is not stable. Would you upload it?

byChen47, 2024-08-16:
Hi @sanguinariojoe ,
This zip file contains cases and dynamic link libraries，you can copy the lib file to OpenFOAM platform  and get it permission，and you can test it by the case 
[deformMesh.zip](https://github.com/user-attachments/files/16636230/deformMesh.zip)

byChen47, 2024-08-16:
when I use the MoorDyn V2.3.3 or low version,the MoorDyn V2 is can use in OpenFOAM，but MoorDyn V2.3.4 or upper  version can't  use in OpenFOAM through foamMooring

sanguinariojoe, 2024-08-16:
That is because on MoorDyn >= 2.3.4 we have a new stationary solver, which is way more powerful, but sometimes a bit slow converging...

You can get the same behavior than before just simply adding the following option to your MoorDyn config:

`1             ICgenDynamic - 0 for stationary solver, 1 for upscaled drag legacy solver`

That is probably fixing this issue. Let me know please

byChen47, 2024-08-16:
> 这是因为在 MoorDyn >= 2.3.4 上，我们有一个新的稳态求解器，它更强大，但有时收敛速度有点慢......
> 
> 您只需将以下选项添加到MoorDyn配置中，即可获得与以前相同的行为：
> 
> `1 ICgenDynamic - 0 for stationary solver, 1 for upscaled drag legacy solver`
> 
> 这可能正在解决这个问题。请让我知道

It's a bit difficult for me， because I don't know where I should add an entire line of commands to that file in MoorDyn

byChen47, 2024-08-16:
Hi @sanguinariojoe ,
I have add the line commands  in the MoorDyn input file ,this ok and the case can run
Thank you very much

closed 2024-08-20 by RyanDavies19