FloatingArrayDesign/MoorDyn #250: How to visualize lines.out

luop21, 2024-08-28:
I am a new user of MoorDyn, anyone know how to visualize lines.out? I want to get a picture

sanguinariojoe, 2024-08-28:
I recommend you to save a VTK file and then use Paraview to visualize it.

Otherwise you can use the API to extract the data yourself and plot with
matplotlib.

Up to you, really. Try to give it a though, check the tests, and let us
know. We can give you further hints if you make your mind

RyanDavies19, 2024-08-28:
@luop21 there's also an option to use [MoorPy](https://github.com/NREL/MoorPy/tree/dev), although the code is under development. If you are savvy with python you can try working with load and the plot functions. The gist of it is you need to load in the system using https://moorpy.readthedocs.io/en/dev/api.html#moorpy.system.System.load, then for each object in the system load the outfile data (ex: `line.loadData(<path to outputs>, "lines", sep = '_') `), and then plot using the plot, plot2d, or some variation of the animation function. This will likely require some modification of things on your side.

sanguinariojoe, 2024-09-06:
Hey @luop21 

Had you taken a look? Do you still need help?

closed 2024-12-12 by sanguinariojoe