FloatingArrayDesign/MoorDyn #254: How to connect two freely floating bodies through Moordyn v2 in Matlab

ferrari700, 2024-10-09:
Hello, I have a question about whether it is possible to connect two freely floating bodies through Moordyn v2 and calculate the cable tension. The parameters and hydrodynamic coefficients of the two floating bodies are calculated by an external program. My input file (txt) in Matlab is as follows: set one floating body as a vessel and the other as coupled, with a ballast block (set as Free) connected in the middle of the cable. Is this feasible? Should we define another floating object as the body and assign parameters in the txt file? Perhaps there are other ways to achieve this goal, thank you very much for your help, sincerely wish you all the best.

--------------------- MoorDyn Input File ------------------------------------
Mooring system for the the floater
----------------------- LINE TYPES ------------------------------------------
TypeName   Diam    Mass/m     EA         BA/-zeta    EI         Cd     Ca     CdAx    CaAx
(name)           (m)     (kg/m)     (N)         (N-s/-)     (N-m^2)   (-)    (-)      (-)           (-)
Seg1	   0.088      47.885    3.13E+08	   -0.8	  0          1     0.025    0           1.2
---------------------- POINT PROPERTIES --------------------------------
ID    Type      X       Y       Z       Mass   Volume  CdA    Ca
(#)   (-)       (m)     (m)     (m)     (kg)    (m^3)    (m^2)  (-)
1     Vessel   -0.1     0       0        0        0            0         0
2     Free     109      0       0       764     9.708      2.5      1
3     Coupled  0.1      0       0       0       0            0         0
---------------------- LINES ----------------------------------------
ID   LineType   AttachA  AttachB  UnstrLen  NumSegs  LineOutputs
(#)   (name)     (#)           (#)              (m)            (-)              (-)
1      Seg1	1	       2                50	           25	              -
2      Seg1	2	       3                50	           25	              -
---------------------- OPTIONS -----------------------------------------
2             writeLog      Write a log file
0.0001         dtM           time step to use in mooring integration (s)
3.0e6         kBot          bottom stiffness (Pa/m)
3.0e5         cBot          bottom damping (Pa-s/m)
1025.0        WtrDnsty      water density (kg/m^3)
1330           WtrDpth       water depth (m)
0.001           dtIC          time interval for analyzing convergence during IC gen (s)
1.0         TmaxIC        max time for ic gen (s)
4.0           CdScaleIC     factor by which to scale drag coefficients during dynamic relaxation (-)
0.001         threshIC      threshold for IC convergence (-)
-------------------------- OUTPUTS --------------------------------
FairTen1
FairTen2
------------------------- need this line --------------------------------------

sanguinariojoe, 2024-10-09:
MoorDyn is meant to solve the mooring lines. The dynamics of the bodies are your responsibility.

You can consider using [OpenFAST](https://github.com/OpenFAST/openfast) to solve your problem (which has already MoorDyn integrated)

ferrari700, 2024-10-09:
I want to calculate the mooring tension between two freely floating objects

sanguinariojoe, 2024-10-09:
Sure. How are you planning to solve the floating objects motion?

ferrari700, 2024-10-09:
I have already built the Cummins equation in Matlab to calculate the motion of floating bodies. I would like to know how to define the position of the mooring point in the input file (txt) if I want to calculate the cable tension between two floating bodies

ferrari700, 2024-10-09:
In other words, I only need to solve how to calculate the mooring line tension between two floating bodies through Moordyn v2, could you provide good advice about the input file? Thank you.

sanguinariojoe, 2024-10-09:
OK, then you have 2 options:

 - Use coupled points, as you are doing now. Then you need to get those 2 points manually solving your body kinematics
 - Create coupled bodies, so you just report the bodies motion and MoorDyn will take care on the rigid body kinematics

A pseudo-matlab code (using points) would look like:

```
p1 = MoorDynM_GetPoint(system, 1)
p2 = MoorDynM_GetPoint(system, 3)
l1 = MoorDynM_GetLine(system, 1)
l2 = MoorDynM_GetLine(system, 2)
t = 0
while 1
    p1_curr = MoorDynM_GetPointPos(p1)
    p2_curr = MoorDynM_GetPointPos(p2)
    p1_next = rigid_body_kin(mybody1,  -0.1 0, 0)
    p2_next = rigid_body_kin(mybody2,  0.1 0, 0)
    u1 = (p1_next - p1_curr) / dt
    u2 = (p2_next - p2_curr) / dt
    MoorDynM_Step(system, [p1_curr p2_curr], [u1 u2], t, dt)

    % Get the forces on the line ends
    f1 = MoorDynM_GetLineNodeTen(l1, 0)
    f2 = MoorDynM_GetLineNodeTen(l2, MoorDynM_GetLineN(l2))

    t += dt
end
```

ferrari700, 2024-10-09:
Thanks a lot. But if I use coupled points, how should I define the floating bodies in the input txt file? The first body is "Vessel", how to define the other body?

sanguinariojoe, 2024-10-09:
"Vessel" and "Coupled" means exactly the same. On that config you are letting MoorDyn know that you will provide the position and velocity of the points 1 and 3.

If you look the example, when I call `MoorDynM_Step()`, I am giving it the position and velocity of both points. The position I get it from Moordyn itself (calling `MoorDynM_GetPointPos()`), but for the velocities you need to use your computed kinematics to know them (I called that `rigid_body_kin()`)

ferrari700, 2024-10-09:
OK, I understand what you mean, thank you very much for your help!

ferrari700, 2024-10-10:
Sorry, I want to know what does the result in the 3 * 1 matrix output by f1 represent? Are they the forces in the x, y, and z directions of the fairlead?

sanguinariojoe, 2024-10-10:
Yes

closed 2024-10-10 by ferrari700

sanguinariojoe, 2024-10-10:
Just to clarify:

Are the forces in the **global** x, y, and z directions **at** the fairlead

ferrari700, 2024-10-10:
The fuction  MoorDynM_Step() also output the force about a 3 * 1 matrix, are they the same result?

sanguinariojoe, 2024-10-10:
Nope... `MoorDynM_Step()` returns the sum of the forces on all the coupled entities (points in your case). I suppose we keep that for backward compatibility, but you can ignore that value.

ferrari700, 2024-10-11:
Thanks a lot. But the force from MoorDynM_GetLineNodeTen() can not be updated with current points, even if I have a new fairlead position, how can I obtain the force on the fairlead?

sanguinariojoe, 2024-10-11:
Of course the force is updated... It is a staggered coupling.

- Get the motions of the fairleads from your floating bodies
- Provide MoorDyn with that info calling to MoorDynM_Step()
- Get the new forces calling to MoorDynM_GetNodeTen()
- Give those forces to your floating bodies an compute the motions for the
next step

You can alternatively use MoorDynM_GetPointForce() if you like that better

ferrari700, 2024-10-12:
I tried to use the MoorDynMGetPointForce() function to calculate the force on the fairlead, but the result is the same as that obtained from MoorDynM_Step (system, [p1_curr p2_curr], [u1 u2], t, dt). Is this reasonable?

RyanDavies19, 2024-10-14:
@ferrari700, 

Yes that is reasonable. MoorDynM_Step returns the forces on the coupled objects in their respective degree of freedom (global frame). MoorDynMGetPointForce() returns the 3 DOF forces on the specified point. Because you have coupled points, then MoorDynM_Step will return the same force as MoorDynMGetPointForce

ferrari700, 2024-10-19:
But there are two fairleads on the coupling body, and I want to calculate the forces on both fairleads separately, rather than the coupling forces acting on the body.

sanguinariojoe, 2024-10-19:
If  `MoorDynM_Step()` is returning a 3x1 array is because there are something wrong on the Matlab code. Look [here](https://github.com/FloatingArrayDesign/MoorDyn/blob/master/wrappers/matlab/MoorDynM_Step.cpp#L55), the output has the same size than the input position (`[p1_curr p2_curr]`).

Matlab is kind of an idiotic language, really prone to mess up with the dimensions. Please, check that `[p1_curr p2_curr]` and `[u1 u2]` are one-dimensional arrays of 6 components each.

ferrari700, 2024-10-23:
So if there are three cable holes, the inputs should be [p1_curr p2_curr p3_curr] and [u1 u2 u3], and the outputs should be [t, F], right?

sanguinariojoe, 2024-10-23:
In your particular case, yes (assuming cable hole = fairlead = coupled point). In general you must provide inputs for all the coupled entities you have in your MoorDyn file.

ferrari700, 2024-10-23:
So the F in the output is a 9 * 1 array, and the force of each fairleads  is a 3 * 1 array, right?

sanguinariojoe, 2024-10-23:
Yep