FloatingArrayDesign/MoorDyn #257: 2 errors running test code MoorDyn python

ewoudVL13, 2024-10-14:
Hello, new to this community. I ran the python example code after pip installing MoorDyn in my python project. Is this V2? (using test/mooring/lines.txt as input file)

I had to make changes on 2 places: 
1: line 12: x = x + moordyn.GetPointPos(point) to x = x + list(moordyn.GetPointPos(point))
  because a tuple can not be added to a list in python

2: line 29:  n_nodes = moordyn.GetLineNumberNodes(line) to n_nodes = moordyn.GetLineN(line)+1 because .GetLineNumberNodes was not found in the library. The use of GetLineN(line)+1 can be a solution for this.

RyanDavies19, 2024-10-14:
@ewoudVL13 Thanks for pointing these out, the example the docs could use updating to bring it in line with current function names. I will be putting together a minor docs update soon and loop this in as part of it.

RyanDavies19, 2024-10-14:
And yes, the moordyn available via pip is MoorDyn V2

closed 2024-10-17 by RyanDavies19

sanguinariojoe, 2024-10-23:
I added the `moordyn.GetLineNumberNodes()` entry on #262