FloatingArrayDesign/MoorDyn #259: moordyn.SaveState does not work in python

ewoudVL13, 2024-10-17:
Hello, tried to run the python program in the "theory and troubleshooting" paragraph of the moordyn.reatherdocs.

The moordyn.SaveState command (line 9 and 16) does not exist in the current python moordyn. I tried just using the moordyn.Save but this gives errors later: NaN for node position 1. 

Also tried to just simulate a line with fixed end points and a current effecting it but can't get it going. Any ideas? The input files for the line and current are ok I think but it seems like it does not simulate the line.

RyanDavies19, 2024-10-17:
Hi @ewoudVL13, thanks for pointing out more types with the docs example. I opened an issue for us to go and update all our examples to the current API. 

As for the errors you are seeing, have you tried a simpler case, like this example: https://moordyn.readthedocs.io/en/latest/drivers.html#python

The Nan you are getting before save state makes me suspect that your configuration is unstable during the initialization. Cn you share your input file? The simulation can become unstable for a number of reasons, the most common are too large of a time step and too small of a line discretization.

ewoudVL13, 2024-10-17:
[chain.txt](https://github.com/user-attachments/files/17415538/chain.txt)
[current_profile.txt](https://github.com/user-attachments/files/17415539/current_profile.txt)
These are my input files.

RyanDavies19, 2024-10-17:
@ewoudVL13,

You have a vertical line in 320m of water that is 380m long, I suspect that is causing the issue. Additionally, a 20m/s current is extremely fast and is likely causing issues as well. 

MoorDyn _might_ be able to handle this if you add `1      ICgenDynamic` to your options list, change your CdScaleIC to 1.0, and reduce your `dtm` time step. However this is a pretty extreme case.

RyanDavies19, 2024-10-17:
@ewoudVL13, 

Glad you got it running, thanks for sharing the driver script. When / if you change to having a coupled point your x and xd lists are going to need to have values for the coupled DOF. Also, if you add FairTen1 to the output list (below the options section), moordyn will write the fairlead tensions to an output file for you rather than needing to print them to the console.

closed 2024-11-26 by RyanDavies19