FloatingArrayDesign/MoorDyn #260: Make sure the docs are up to date with the current API

RyanDavies19, 2024-10-17:
Hi @sanguinariojoe, 

If you are opening up a PR soon can you make sure the docs examples are up to date with the current API (See #257, #259)?

I figure it'll be quicker for you to do it because you set those examples up.

sanguinariojoe, 2024-10-18:
Regarding #259 (affecting [the initialization section](https://moordyn.readthedocs.io/en/latest/initialization.html)), the API entry is already there, on the dev branch though:

https://github.com/FloatingArrayDesign/MoorDyn/blob/dev/wrappers/python/moordyn/moordyn.py#L501

The problem is that the default MoorDyn readthedocs points towards dev, not master.

I will check the API and the docs anyway

RyanDavies19, 2024-10-18:
@sanguinariojoe Thanks!

sanguinariojoe, 2024-10-23:
Fixing on #262

closed 2024-12-12 by sanguinariojoe