FloatingArrayDesign/MoorDyn #263: Output error in time domain?

ewoudVL13, 2024-10-21:
Hello, I successfully modeled a buoy in a current. I would like to have the outputs (position of buoy and tension in line) from [0-50] in the time domain but it only gives the ones at 0 and 0.5 (the dt in the program). What is going on? 

This is a step to when I implement waves and the tension and position will change in time due to the waves. 

[buoy.log](https://github.com/user-attachments/files/17463270/buoy.log)
[buoy.txt](https://github.com/user-attachments/files/17463271/buoy.txt)
[buoy_program.txt](https://github.com/user-attachments/files/17463272/buoy_program.txt)
[current_profile.txt](https://github.com/user-attachments/files/17463273/current_profile.txt)
The used programs are here.
Thanks in advance, Ewoud

ewoudVL13, 2024-10-21:
It works @RyanDavies19 THX! 
If I want the VTK file like a video that shows how the buoy moves do I just put that in the for loop or is there a specific way to do that?

ewoudVL13, 2024-10-21:
@RyanDavies19 or colleagues, My buoy has 6DOF now its defined as free. Is it also possible to give it 3DOF: only translations, no rotations?

RyanDavies19, 2024-10-21:
Hi @ewoudVL13, no you cannot explicitly restrict the DOF of rods. There are a few ways around this depending on what you want. One option is to replace the rod with a point, that has the volume of the submerged part of the buoy and the mass of the whole buoy. This will give you the hydrostatic properties with no rotations, but you will loose the hydrodynamics with the rod.

The other option is to attach the rod to a body, set the rod mass to zero, and assign the body the mass of the buoy. From there you can set the body moment of inertia to very high (10^12), which will effectively restrict all rotations. This approach retains the hydrostatics and dynamics of the rod, but takes a bit more to set it up.

As for the VTK stuff, that's not my area of expertise so I can't be much help. @sanguinariojoe might have an answer.

sanguinariojoe, 2024-10-23:
Hey @ewoudVL13

Sure you can produce an animation like a video. Just call to `moordyn.SaveVTK()` inside the loop. If you want Paraview to understand that those files are actually an animation you should rename your files with the "prefix.{i:04d}.vtm" name format. e.g.

```
dt = 0.5
time = np.arange(0,50,dt)
for i, t in enumerate(time):
    moordyn.Step(system, x, xd, t, dt)
    moordyn.SaveVTK(system, f"mooring.{i:04d}.vtm")
```

closed 2024-12-12 by sanguinariojoe