FloatingArrayDesign/MoorDyn #264: Regular wave input file

ewoudVL13, 2024-10-22:
Hello all, 
I have created a buoy and put in a regular wave climate. I used WaveKin=2 (FFT in a regular grid) for this. 
The regular wave climate is: amplitude 5m, T=8.33s == f=0.12Hz, (wave length 103m). I did not get a clean sinus as I would have expected for the motion of the buoy. Is this because of the interpolation between 0 and the 0.12 f in the wave file?
The wave frequency file is added below.   
 [wave_frequencies.txt](https://github.com/user-attachments/files/17478287/wave_frequencies.txt)
Does this file generate a regular wave system?

If I would go to a irregular wave profile based on JONSWAP spectrum. See input file below. Would this work?
[wave_frequencies.txtirregular waves.txt](https://github.com/user-attachments/files/17478314/wave_frequencies.txtirregular.waves.txt)

I created the wave grid as according to the manual. This creates a general area where the buoy and attachments are active and wave interaction is possible if I understand correctly.

THX in advance, Ewoud

RyanDavies19, 2024-10-22:
Hi @ewoudVL13,

If you're just looking for a simple sine wave, WaveKin=3 is probably going to be easier and more accurate. WaveKin=2 struggles with low resolutions spectra (like what you provided). Did the JONSWAP file you provided give better results? 

It sounds like you have the wave grid right. You want to make sure the grid encompasses everywhere that you want to simulate waves interacting with your device. For example, if your waves are all along the X axis you will want high resolution in the x direction of the grid, low resolution in the y direction (to save memory), and z resolution that captures the full wave height (provided you are not including currents).

RyanDavies19, 2024-10-22:
@ewoudVL13 I would think your JONSWAP would be high enough resolution but the best bet is to just try it out and see if the results look right.

closed 2024-11-19 by RyanDavies19