FloatingArrayDesign/MoorDyn #268: Big discrepencys when comparing Moordyn V2 to Orcaflex

ewoudVL13, 2024-11-06:
Hello everybody,

I'm currently working on a comparison study with MoorDyn V2 and Orca Flex. The base model I'm using is a buoy connected to the seabed. See input data below.  
[image]
But when comparing the two the buoy does not get displaced as much as in Orca Flex and the tension in the chain line/chain is lower (+/-7m and1500N less). In this comparison the buoy is modeled as a rod in the input data file and do I look at the first 150s. The input files for the current are shown below.
[buoy_wave.txt](https://github.com/user-attachments/files/17648026/buoy_wave.txt)
[current_profile.txt](https://github.com/user-attachments/files/17648028/current_profile.txt)
It looks like there is not enough drag?

I then went on to model the buoy as a body but withe input file below the body starts flying out of the water and levitates above the surface.
[buoy.txt](https://github.com/user-attachments/files/17648066/buoy.txt)
What is wrong in the input?

What is your view on this?
Tanks in advance for the help. The final results and findings of my study will be shared with the community.
Greetings Ewoud

RyanDavies19, 2024-11-06:
Hi @ewoudVL13 and @AlexWKinley, 

I was just working on a response, @AlexWKinley beat me to it. I second everything Alex said, the difference between drag approaches in OrcaFlex can be typically be overcome by setting the two drag force equations equal and working through some algebra to get equivalent coefficients. For example, this is how we currently approximate buoyancy element contributions in dynamic power cables. 

The reason the approach purely using a body does not work is that we do not account for partial submergence of bodies when calculating hydrostatic and hydrodynamic forces. In other words the body is always considered submerged. A form of hybrid approach that I have successfully used in the past was modeling the kinematics as a body and the hydrostatic and dynamic forces as a rod. This means the body has no volume, but it does contain the mass and inertia values. Then the rod has the drag coefficients and volume, but zero mass. You have to be careful with this approach, as the body position needs to be located at the center of mass of the buoy, otherwise the kinematics will be off.

RyanDavies19, 2024-11-06:
@AlexWKinley, we have been discussing some ideas for horizontal submergence on our end but haven't fully worked through them yet. I have been seeing a number of results showing issues with the rod hydrostatics when the rod is partially submerged near horizontal on the MD-F side, but there are other OpenFAST modules that have allowed for work-arounds to model cylindrical objects behavior.

ewoudVL13, 2024-11-06:
@RyanDavies19 and @AlexWKinley  My buoy falls flat (horizontal) in the current OrcaFlex and MoorDyn calculations. Is it then not strange that there is such a difference in values? 

Thx for the answers I 'll have some results tomorrow.

RyanDavies19, 2024-11-06:
> @RyanDavies19 and @AlexWKinley My buoy falls flat (horizontal) in the current OrcaFlex and MoorDyn calculations. Is it then not strange that there is such a difference in values?
> 
> Thx for the answers I 'll have some results tomorrow.

@ewoudVL13, interesting. When running your input files you provided the buoy stays vertical for me. Regardless, it will be interesting to see results.

ewoudVL13, 2024-11-07:
@RyanDavies19 and @AlexWKinley The webhelp of OrcaFlex tells me that for a spar buoy 6DOF(https://www.orcina.com/webhelp/OrcaFlex/Content/html/6Dbuoys,Sparbuoyandtowedfishdragslam.htm#StackedBuoyDrag) the drag area in axial direction is equal to the area of the circle. 

So what have I done:  put the axial area to this value in orcaflex with the corresponding drag factor (Cd=1.17), calculate the p value in OrcaFlex (via dry length of the buoy) and us it to calculate the adapted Cd for use in MoorDyn. 
Result: the moordyn and OrcaFlex corelate with a 1-2%difference. (Moordyn always has the lowest value)

I still find it strange that the buoy stays upright with you @RyanDavies19. She falls flat in MD and OF in my simulations.
Next up: I will look how the buoy behaves in the waves. I think that the likelihood of  issues with the buoyancy due to the fact that the buoy lays flat will be larger but will see. Also the p factor can change there.

AlexWKinley, 2024-11-07:
Glad to hear you improved the agreement. 

I suspect you'll find that in waves things are more hairy, both due to the errors in buoyancy calculation, and some suspicious looking (to me) wave forcing on rods in MoorDyn. Specifically MoorDyn includes a force that looks like this, on the rod ends
```cpp
// Froude-Krylov force
Pd[i] += VOF[i] * Area * PDyn[i] * q;
// VOF[i] is segment submergence
// Area is just pi * r^2
// PDyn is wave dynamic pressure at the node
// q is the vector describing the rod orientation
```
This is on top of the added mass inertial force which is also included. I don't have a good grasp on the underlying theory, but as far as I know OrcaFlex doesn't include a force like this.

All of this is to say, definitely keep us updated on your work. 

Additionally, if you feel comfortable sharing your orcaflex model, and can save it as a .yml file, that potentially can help find discrepancies. But it's up to you.

ewoudVL13, 2024-11-07:
Indeed when running through an irregular wave it lost buoyancy. A regular wave climate was more manageable but dit not correlate to the OF simulation (+/-10% difference). 
My current simulating OF file.
[%YAML 1.1.txt](https://github.com/user-attachments/files/17663986/YAML.1.1.txt)

I 'm also looking at creating a buoy with a movable location for center of mass i can choose my self with a bodie fixed to a rod. 
I created an input file but it is fixed I think. Do you or @RyanDavies19 have any ideas or examples you could share?  
[buoy_mass.txt](https://github.com/user-attachments/files/17664381/buoy_mass.txt)

Also, is it possible to connect 2 rods to each other (fixed connection)? Did not find anything in the manual. (could be handy to create a buoy with a stepped design or also artificially lowering the center of mass but not needing to put in the inertias) 
THX in advance!

RyanDavies19, 2024-11-07:
@ewoudVL13 to your last two points: First, your `buoy_mass.txt` file looks pretty good. You just need to change the body attachment to free and the rod attachment to body1:
```
---------------------- BODIES ---------------------------------------
ID   Attachment  X0     Y0    Z0     r0      p0     y0     Mass  CG*   I*      Volume   CdA*   Ca*
(#)   (word)     (m)    (m)   (m)   (deg)   (deg)  (deg)   (kg)  (m)  (kg-m^2)  (m^3)   (m^2)  (-)
1     Free        30     0     0     0       0      0      1200   0  7905.63|7905.63|93.615  0       0      0
--------------------- RODS ----------------------------------------
ID   RodType  Attachment  Xa    Ya    Za    Xb    Yb    Zb   NumSegs  RodOutputs
(#)  (name)   (word/ID)   (m)   (m)   (m)   (m)   (m)   (m)  (-)       (-)
1     Buoy     Body1       30    0    -4.43  30     0   4.435  8         -
```

This makes the rod end A and B locations defined relative to the body's position, which is defined as 0,0,0. If for example you wanted the mass and MOI about a point 1/4th the total length from the bottom of the buoy, you would define Za as -2.215 and Zb as 6.645, and the body z position (Z0) as -2.215. 

Second, you can attach two rods rigidly, using a body. The body can be thought of as the frame that everything gets attached to (everything attached to a body is rigidly coupled except for pinned rods). You would define the body position, and then the two rods positions in the body's reference frame. Something like:
```
---------------------- BODIES ---------------------------------------
ID   Attachment  X0     Y0    Z0     r0      p0     y0     Mass  CG*   I*      Volume   CdA*   Ca*
(#)   (word)     (m)    (m)   (m)   (deg)   (deg)  (deg)   (kg)  (m)  (kg-m^2)  (m^3)   (m^2)  (-)
1     Free        30     0     0     0       0      0      1200   0  7905.63|7905.63|93.615  0       0      0
--------------------- RODS ----------------------------------------
ID   RodType  Attachment  Xa    Ya    Za    Xb    Yb    Zb   NumSegs  RodOutputs
(#)  (name)   (word/ID)   (m)   (m)   (m)   (m)   (m)   (m)  (-)       (-)
1     Buoy1     Body1       30    0    0      30     0   4.435  8         -
2     Buoy2     Body1       30    0    -4.43  30     0   0      8         -
```

As for the buoy not falling over, I am using the exact input files you shared, with no changes. My coupling timestep is 0.0001, the same as dtM. I am running MoorDyn 2.3.7, the current release.

ewoudVL13, 2024-11-18:
@AlexWKinley is it possible to share a email address and some more of your findings on the troubles with moordyn, buoys, waves and orcaflex. I 'm currently working on a mussel line model for my thesis and am running into troubles to get them converging.

sanguinariojoe, 2024-12-12:
I close this since apparently they are having a parallel conversation

closed 2024-12-12 by sanguinariojoe