FloatingArrayDesign/MoorDyn #270: How to run the case?

LJM-97, 2024-11-10:
Hi everyone @sanguinariojoe @mattEhall,

Nice of you to share the Moordyn. 

- I am new to this solver and I have some confusion about using it. I have successfully compiled the Linux-fortran version, **_but how do I run it_**? 

- In master/bench, there are only mooring line settings, does this mean that **_Moordyn_** can only be called by other programs?

There doesn't seem to be any description in the documentation either. It seems no executable program, anyone who can help me point out the solution to use it would be greatly appreciated! 

Best regards!

RyanDavies19, 2024-11-11:
Hi @LJM-97, check out this example in the documentation: https://moordyn.readthedocs.io/en/latest/drivers.html#fortran The bare minimum requirement, after initialization, is to call MoorDyn_Step for each time step. The example above iterates one time step. 

In short, MoorDyn is a library that requires an external program to call it. We have wrappers for Python, Matlab, and Fortran.

LJM-97, 2024-11-13:
> Hi @LJM-97, check out this example in the documentation: https://moordyn.readthedocs.io/en/latest/drivers.html#fortran The bare minimum requirement, after initialization, is to call MoorDyn_Step for each time step. The example above iterates one time step.
> 
> In short, MoorDyn is a library that requires an external program to call it. We have wrappers for Python, Matlab, and Fortran that allow for this.

Thanks for your answer. My other question is, can the lumped-mass model in the current version of MoorDyn be used to simulate the aquaculture cage dynamics?

Best

AlexWKinley, 2024-11-13:
Not to butt in, but I think I can potentially provide some insight here.

In terms of creating a "model runner" executable, since MoorDyn is designed with coupling in mind, you have to provide your own code to actually call MoorDyn. This can be done in a number of languages (including C/C++, Python, and MATLAB). 

To see some examples of running MoorDyn models, you can look through the `tests` folder in this repo.
In particular https://github.com/FloatingArrayDesign/MoorDyn/blob/master/tests/tripod.cpp is basically the simplest example for running a MoorDyn model from C/C++.
And if you're using python
https://github.com/FloatingArrayDesign/MoorDyn/blob/master/tests/test_minimal.py
and
https://github.com/FloatingArrayDesign/MoorDyn/blob/master/tests/test_generator.py
would be worth looking through.

In terms of whether MoorDyn is applicable for simulating aquaculture cage dynamics, the answer is probably it depends. The fundamental flexible dynamics of a flexible chain or rope are handled well by MoorDyn. But things like fluid structure interaction are not supported in any direct way, and MoorDyn doesn't always provide the capabilities to approximate them. It can be possible to couple external programs with MoorDyn (potentially including fluid solvers). But whether that meets your needs is very use case specific.

LJM-97, 2024-11-14:
> I'm not entirely sure what you mean by "aquaculture cage dynamics". MoorDyn can certainly simulate the mooring lines for aquaculture systems, and it has some capability to simulate things like buoys and other surface piercing objects. However, it really isn't set up to simulate the forcing on complex structures themselves. Typically this part of the simulation would be handled by an external program, while MoorDyn handles the mooring system.

Dear Alex,

What I want to express is whether Moordyn can be used to calculate the hydrodynamics of net coats in an aquaculture cage. In some cases, the net coat is applied to a lumped-mass model. If it is possible, the net coat after clustering is able to be represented by this model.

Best

LJM-97, 2024-11-14:
> Not to butt in, but I think I can potentially provide some insight here.
>
> In terms of creating a "model runner" executable, since MoorDyn is designed with coupling in mind, you have to provide your [...]

Hi Alex,
With your detailed description, I think I understand how MoorDyn is used, thanks for your reply!

closed 2024-11-19 by RyanDavies19